energia/Energia #934: Add Wire.setModule() to the Wire Library (I²C) reference page

rei-vilo, 2016-10-05:
Add `Wire.setModule()` to the [Wire Library (I²C) reference page](http://energia.nu/reference/wire/).

duyka0110, 2016-10-06:
excuse me, what is the parameter? is it the address of the I2C pins?

rei-vilo, 2016-10-06:
No, it is the port number. Please refer to the [pins map](http://energia.nu/wordpress/wp-content/uploads/2014/01/LaunchPads-LM4F-TM4C-—-Pins-Maps-11-32.jpeg).

duyka0110, 2016-10-06:
for example the I2C0. i need 2 setModule, one for the SCL and one for the SDA, right?

rei-vilo, 2016-10-06:
No, you need only one `setModule` for I²C port 0. You're selecting the I²C port, not the pins. The I²C port knows which pins to use.

duyka0110, 2016-10-06:
okay thank you very much

hemangjoshi37a, 2017-05-03:
I am using msp430 launchpad  and i want to change spi module from UCB0 to UCA0. What should i do???

hemangjoshi37a, 2017-05-03:
In case If i want to change i2c communication then which pins will be availabe for i2c for wire.setModule(1) or wire.setModule(2) ?? means which pin is scl and sda for different modules.??

rei-vilo, 2017-05-03:
Feel free to 

* Have a look at the data sheet of the exact MSP430 you're using, and 
* Switch to Code Composer Studio which provides full access to all features of the MSP430 you're using.

StefanSch, 2017-05-03:
the pinmaps on energia,nu also shows this information as implemented in the Energia lib.
e.g.: http://energia.nu/pin-maps/guide_msp430fr5969launchpad/
shows behind SCL (x) where x is the id used for setModule

StefanSch, 2017-05-04:
To change SPI from UCB0 to UCA0 you have to hack the energia source files, start with this files:
.\hardware\energia\msp430\variants\MSP-EXP430G2553LP\pins_energia.h
.\hardware\energia\msp430\libraries\SPI\utility\eusci_spi.cpp

hemangjoshi37a, 2017-05-04:
@rei-vilo , i am not  using code composer studio.
@duyka0110 , Okay thax bro...

robertinant, 2017-05-04:
@hemangjoshi37a please use the Energia forum @ http://forum.43oh.com/forum/119-energia/ for questions. Github issues as named are only for issues and not for questions. If it is determined that after you post on 43oh, your question turns out to have to do with a bug, then you can post the bug here on github.

hemangjoshi37a, 2017-05-05:
@robertinant  
Okay bro sure...

robertinant, 2017-05-05:
I have reopened it since this is still a valid issue. Wire.setModule() is currently not documented on the Energia reference page. You used this thread to ask a question hence I requested you to post it in the forum instead.