energia/Energia #938: ENERGIA 1610e18 no upload to MSP430XXXX

steeley, 2016-10-29:
This version produces an error when trying to upload  a sketch(launchpad 1.5 board):

An error occurred while uploading the sketch
dyld: Library not loaded: /opt/local/lib/libusb-0.1.4.dylib
  Referenced from: /Users/ian/Downloads/Energia.app/Contents/Java/hardware/tools/msp430/bin/mspdebug
  Reason: image not found

Sketches can be uploaded to this board if I use an old Energia  (101e015).

rei-vilo, 2016-10-29:
Which OS please?

steeley, 2016-10-29:
Sorry forgot to say: OSX 10.11.6

rei-vilo, 2016-10-29:
Which MSP430 exactly? 

I have no issue uploading on macOS Sierra.

steeley, 2016-10-29:
MSP 430g2553. Why is it looking for an old libusb library?? I can upload to a new 432 board with the new Energia, so that works.

rei-vilo, 2016-10-29:
MSP430G2553 uses the old `mspdebug` while MSP432 relies on `DSLite`.

Both work for me.

steeley, 2016-10-29:
But this does not explain why the older version of Energia DOES work ok and the new version fails!

rei-vilo, 2016-10-29:
As said before, uploading to the MSP430G2553 works fine for me with Energia 18 on macOS Sierra.

steeley, 2016-10-29:
/Users/ian/Downloads/Energia.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Users/ian/Downloads/Energia.app/Contents/Java/hardware" -hardware "/Users/ian/Library/Energia15/packages" -tools "/Users/ian/Downloads/Energia.app/Contents/Java/tools-builder" -tools "/Users/ian/Downloads/Energia.app/Contents/Java/hardware/tools/msp430" -tools "/Users/ian/Library/Energia15/packages" -built-in-libraries "/Users/ian/Downloads/Energia.app/Contents/Java/libraries" -libraries "/Users/ian/Documents/Energia/libraries" -fqbn=energia:msp430:MSP-EXP430G2553LP -ide-version=10610 -build-path "/var/folders/t0/vvzmr4l91jvdq7b2nkxb3cgm0000gn/T/build934c897db08bb52eb726f454b68c7bff.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/ian/Downloads/Energia.app/Contents/Java/examples/01.Basics/Blink/Blink.ino"
/Users/ian/Downloads/Energia.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Users/ian/Downloads/Energia.app/Contents/Java/hardware" -hardware "/Users/ian/Library/Energia15/packages" -tools "/Users/ian/Downloads/Energia.app/Contents/Java/tools-builder" -tools "/Users/ian/Downloads/Energia.app/Contents/Java/hardware/tools/msp430" -tools "/Users/ian/Library/Energia15/packages" -built-in-libraries "/Users/ian/Downloads/Energia.app/Contents/Java/libraries" -libraries "/Users/ian/Documents/Energia/libraries" -fqbn=energia:msp430:MSP-EXP430G2553LP -ide-version=10610 -build-path "/var/folders/t0/vvzmr4l91jvdq7b2nkxb3cgm0000gn/T/build934c897db08bb52eb726f454b68c7bff.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/ian/Downloads/Energia.app/Contents/Java/examples/01.Basics/Blink/Blink.ino"
Using board 'MSP-EXP430G2553LP' from platform in folder: /Users/ian/Downloads/Energia.app/Contents/Java/hardware/energia/msp430
Using core 'msp430' from platform in folder: /Users/ian/Downloads/Energia.app/Contents/Java/hardware/energia/msp430
Detecting libraries used...
"/Users/ian/Downloads/Energia.app/Contents/Java/hardware/tools/msp430/bin/msp430-g++" -c -g -O2  -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=msp430g2553 -DF_CPU=16000000L -DARDUINO=10610 -DENERGIA=10610 -DENERGIA_MSP-EXP430G2553LP -DENERGIA_ARCH_MSP430 -I/Users/ian/Downloads/Energia.app/Contents/Java/hardware/tools/msp430/include  "-I/Users/ian/Downloads/Energia.app/Contents/Java/hardware/energia/msp430/cores/msp430" "-I/Users/ian/Downloads/Energia.app/Contents/Java/hardware/energia/msp430/variants/MSP-EXP430G2553LP" "/var/folders/t0/vvzmr4l91jvdq7b2nkxb3cgm0000gn/T/build934c897db08bb52eb726f454b68c7bff.tmp/sketch/Blink.ino.cpp" -o "/dev/null"
"/Users/ian/Downloads/Energia.app/Contents/Java/hardware/tools/msp430/bin/msp430-g++" -c -g -O2  -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=msp430g2553 -DF_CPU=16000000L -DARDUINO=10610 -DENERGIA=10610 -DENERGIA_MSP-EXP430G2553LP -DENERGIA_ARCH_MSP430 -I/Users/ian/Downloads/Energia.app/Contents/Java/hardware/tools/msp430/include  "-I/Users/ian/Downloads/Energia.app/Contents/Java/hardware/energia/msp430/cores/msp430" "-I/Users/ian/Downloads/Energia.app/Contents/Java/hardware/energia/msp430/variants/MSP-EXP430G2553LP" "/var/folders/t0/vvzmr4l91jvdq7b2nkxb3cgm0000gn/T/build934c897db08bb52eb726f454b68c7bff.tmp/sketch/Blink.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/Users/ian/Downloads/Energia.app/Contents/Java/hardware/tools/msp430/bin/msp430-g++" -c -g -O2  -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=msp430g2553 -DF_CPU=16000000L -DARDUINO=10610 -DENERGIA=10610 -DENERGIA_MSP-EXP430G2553LP -DENERGIA_ARCH_MSP430 -I/Users/ian/Downloads/Energia.app/Contents/Java/hardware/tools/msp430/include  "-I/Users/ian/Downloads/Energia.app/Contents/Java/hardware/energia/msp430/cores/msp430" "-I/Users/ian/Downloads/Energia.app/Contents/Java/hardware/energia/msp430/variants/MSP-EXP430G2553LP" "/var/folders/t0/vvzmr4l91jvdq7b2nkxb3cgm0000gn/T/build934c897db08bb52eb726f454b68c7bff.tmp/sketch/Blink.ino.cpp" -o "/var/folders/t0/vvzmr4l91jvdq7b2nkxb3cgm0000gn/T/build934c897db08bb52eb726f454b68c7bff.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
"/Users/ian/Downloads/Energia.app/Contents/Java/tools-builder/ctags/5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/t0/vvzmr4l91jvdq7b2nkxb3cgm0000gn/T/build934c897db08bb52eb726f454b68c7bff.tmp/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/Users/ian/Downloads/Energia.app/Contents/Java/hardware/tools/msp430/bin/msp430-g++" -c -g -O2  -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=msp430g2553 -DF_CPU=16000000L -DARDUINO=10610 -DENERGIA=10610 -DENERGIA_MSP-EXP430G2553LP -DENERGIA_ARCH_MSP430 -I/Users/ian/Downloads/Energia.app/Contents/Java/hardware/tools/msp430/include  "-I/Users/ian/Downloads/Energia.app/Contents/Java/hardware/energia/msp430/cores/msp430" "-I/Users/ian/Downloads/Energia.app/Contents/Java/hardware/energia/msp430/variants/MSP-EXP430G2553LP" "/var/folders/t0/vvzmr4l91jvdq7b2nkxb3cgm0000gn/T/build934c897db08bb52eb726f454b68c7bff.tmp/sketch/Blink.ino.cpp" -o "/var/folders/t0/vvzmr4l91jvdq7b2nkxb3cgm0000gn/T/build934c897db08bb52eb726f454b68c7bff.tmp/sketch/Blink.ino.cpp.o"
Compiling libraries...
Compiling core...
[... cut ...]
Global variables use 20 bytes (3%) of dynamic memory, leaving 492 bytes for local variables. Maximum is 512 bytes.
/Users/ian/Downloads/Energia.app/Contents/Java/hardware/tools/msp430/bin/mspdebug rf2500 --force-reset prog /var/folders/t0/vvzmr4l91jvdq7b2nkxb3cgm0000gn/T/build934c897db08bb52eb726f454b68c7bff.tmp/Blink.ino.hex 
dyld: Library not loaded: /opt/local/lib/libusb-0.1.4.dylib
An error occurred while uploading the sketch
  Referenced from: /Users/ian/Downloads/Energia.app/Contents/Java/hardware/tools/msp430/bin/mspdebug
  Reason: image not found

rei-vilo, 2016-10-29:
See the instructions at [Setup Energia on Mac OS X](http://energia.nu/guide/guide_macosx/)

> **MSP-EXP430G2 and MSP-EXP430FR5739 LaunchPad**
>
* Download the LaunchPad drivers for Mac OS X: LaunchPad CDC drivers zip file for Mac OS X. This driver has been signed and works on Mac OS X 10.x (Yosemite).
* Unzip and double click MSP430LPCDC 1.0.3b-Signed.pkg
* Follow the instructions.
>

steeley, 2016-10-29:
Tried this, exactly the same result...
Old version of Energia uploads, new one doesn't. Same error message.

rei-vilo, 2016-10-29:
Strange. Please open a **Terminal** window and launch 

```
$ /Applications/IDE/Energia.app/Contents/Java/hardware/tools/msp430/bin/mspdebug --version
MSPDebug version 0.23 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2015 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Chip info database from MSP430.dll v3.3.1.4 Copyright (C) 2013 TI, Inc.
```

steeley, 2016-10-29:
/Energia.app/Contents/Java/hardware/tools/msp430/bin/mspdebug --version
-bash: ./Energia.app/Contents/Java/hardware/tools/msp430/bin/mspdebug: No such file or directory

rei-vilo, 2016-10-29:
Open the folder Energia.app and go down to `mspdebug`. 

If necessary, reinstall Energia.

steeley, 2016-10-29:
Reinstalling Energia makes absoulutely no difference.

steeley, 2016-10-30:
OK.... I downloaded the previous version of Energia (17) and copied mspdebug and libmsp430.dylib
into Energia18.app/Contents/Java/hardware/tools/msp430/bin/mspdebug and it now works ok.

So the problem is definitely with the version of mspdebug in 018.

mspdebug --version now shows:
"MSPDebug version 0.22 - debugging tool for MSP430 MCUs"

robertinant, 2016-11-16:
Attached is the right version of mspdebug that does not rely on libusb-0.1.4.dylib.
[mspdebug.zip](https://github.com/energia/Energia/files/595237/mspdebug.zip)

1. unzip
2. Control click Energia.app and select "Show Package Content"
3. Browse to Contents/Java/hardware/tools/msp430/bin/
4. Drag the unzipped mspdebug into that folder

cyoung, 2016-12-04:
I got the same error as @steeley under the same conditions (except on macOS Sierra - 10.12.1).

The mspdebug.zip fix by @robertinant resolved the issue for me.

wchundak, 2016-12-16:
Had same issue with Energia R18 and libusb-0.1.4.dylib on MacOS Sierra 10.12.2. Thanks @robertinant for fix!

worayoot, 2016-12-21:
Thanks to @robertinant ^-^

rossb93, 2016-12-27:
Same issue on Sierra 10.12.2. @robertinant fix did the trick

rpilotte, 2017-03-10:
Still the same issue on Sierra 10.12.3 with Energia 1.6.10E18.
Thanks to @robertinant for the workaround!

MicroJoe, 2017-03-15:
Same issue on El Captain 10.11.6 with Energia 1.6.10E18. The @robertinant workaround worked too.

indirtwetrust, 2017-06-30:
Thank You  @robertinant !!!!   I made an account just to comment and thank you for this solution.  I've been loading Energia sketches through CCS, kind of a pain.

buffstop, 2017-07-12:
Thanks to @robertinant

peterjbaylis, 2018-06-26:
thanks  @robertinant your fix worked for me on OSX 10.13.5

mringwal, 2018-07-02:
thanks for providing a workaround. 

Following up: It would be great to add the solution to the Mac install guide or at least link to this thread in guide. Otherwise, every new user will: follow the instructions, fail, and haves to rely on Google to find a solution for me.