energia/Energia #944: Location of WiFi UDP library code in the directory?

wengchen1993, 2016-12-08:
Hi, I am fairly new to Energia and want to try implement the UDP multicast function in the Wifi UDP library. I downloaded Energia from the official website but have no clue where the Wifi UDP library .h file and .cpp files are located. Can someone please kindly point it out for me? 

(Only WiFi UDP files I found are in the energia-1.6.10E18/hardware/energia/msp430/libraries/WiFi directory and editing them doesn't have any effect. Currently running Energia on CC3200)

StefanSch, 2016-12-08:
Have a look into this thread it lists path for the various operating systems:
https://github.com/energia/cc3200-core/issues/5#issuecomment-262907040

rei-vilo, 2016-12-08:
Feel free to post your question at the 43oh.com forum.

closed 2016-12-08 by rei-vilo