energia/Energia #945: Allow access to pins not on Launchpad

Tejp1996, 2016-12-10:
Microcontrollers such as the MSP are very commonly embedded into custom PCBs. Energia is a great platform for debugging/testing, and even for full deployment. However, because only launchpad pins are accessible, that makes a lot of pins unusable for these things. 

In my case, the MSP430F5529 has tons of pins that are not accessible on the launchpad headers that are being used on a custom PCB. Maybe I'm missing something?

rei-vilo, 2016-12-10:
Energia focuses on the LaunchPads and BoosterPacks. 

For using the full potential of the MCUs, please consider [Code Composer Studio](http://www.ti.com/tool/ccstudio).

closed 2016-12-10 by rei-vilo

Tejp1996, 2016-12-10:
They could easily be added to the list of launchpad pins in addition. CCS is not as easy to prototype with as Energia. Energia doesn't have to shift focus from LaunchPads. Im sure many others will find this useful.

rei-vilo, 2016-12-10:
Please see Getting Started Guide for Developers #907.