energia/Energia #947: Accessing msp430's upper 64KB memory banks

dazhbog, 2016-12-15:
Hi there!

I have been trying for a while to use the upper bank C and bank D memory areas on my MSP430F5529. Since our compiler is using only the lower 64KB I assume it is safe to do so without overriding any program code.

I have a few questions however:

- I tried the built-in flash read and write examples but i was unable to write to the upper part of the memory. My MSP keeps resetting. Is it because energia is using the small model and all the pointers are 16bit? Will the following long pointer be cropped to 16bits once compiled? Or am I doing something wrong?
```C
unsigned long * Flash_ptr;
Flash_ptr = (unsigned long *)0x1C400;
*(unsigned int *)Flash_ptr = 0;    // write some data  
```

- Can I use inline assembly to do so? I read that I have to have the --core=430X flag enabled? Can I do something like this currently? 
```C
void storeToFullAddr(uint32_t pDst, uint8_t data)
{
  asm(" push r2"::);
  asm(" dint"::);
  asm(" push r15"::);
  asm(" push r14"::);
  asm(" popx.a r15"::);
  asm(" movx.b r13, 0(r15)"::);
  asm(" pop r2"::);
}
```

- Do you have any examples or tips on how to read and write a byte in the extended flash memory?

Thanks!

spirilis, 2016-12-15:
For Wolverine (FRAM) usage, take a look at this- http://forum.43oh.com/topic/5474-energia-and-wolverine-tips/ - including my "HighMem" library which employs DMA to copy to/fro buffers within the lower-64KB space.

dazhbog, 2016-12-15:
Thanks for the example. Looking at the HighMen lib i can see that this is very close to what I am looking for... but im stuck at actually implementing it :(

I struggle to pass the large unsigned long C pointer into a the generic registers r12-r15 and also how to return the data from assembly to C. 

I can see many ways that it can be done, but i cant get any to work.. For example in this document (http://www.ti.com/lit/an/slaa376/slaa376.pdf) they show the intrinsic functions that I need, but i dont know how to encapsulate that with C, in Energia.

dazhbog, 2016-12-15:
OK I think I might have solved it. Borrowing from the naming convention from the intrinsics I made the following 2 functions.

```C
void __data20_write_char(unsigned long addr, unsigned char data)
{
  asm(" push   r15"::);           // push MSB part of addr argument
  asm(" push   r14"::);           // push LSB part of addr argument
  asm(" popx.a r15"::);           // load dat 20bit address
  asm(" movx.b r13, 0x0(r15)"::); // move the data argument to the location r15
  asm(" nop"::);                  // be useless for a bit
}

unsigned char __data20_read_char(unsigned long addr)
{
  asm(" push   r15"::);            // push MSB part of addr argument
  asm(" push   r14"::);            // push LSB part of addr argument
  asm(" popx.a r15"::);            // load dat 20bit address
  asm(" movx.b 0x0(R15), R12"::);  // move contents of the R15 location to R12
  asm(" push.w R12"::);            // not sure how to do this but here trying to move the result 
  asm(" pop.w  R15"::);            // back into r15 so that the char can be returned.
  asm(" nop"::);                   // if i remove this it doesnt work..
  asm(" ret"::);                   // return the contents of r15
}
```

I shall test more and report..

Closing for now.

closed 2016-12-15 by dazhbog