energia/Energia #949: MSP430-CC1310

github4f, 2016-12-17:
Hi-is there any library for msp430 to use cc1310/cc1350 as its network processor? Thank you

rei-vilo, 2016-12-17:
The CC1310 and CC1350 LaunchPads are supported by Energia 18. Both MCUs bring a Cortex-M3 plus the sub-1 GHz radio.

See https://github.com/energia/cc13xx-core and https://github.com/energia/EasyLink.

closed 2016-12-17 by rei-vilo

github4f, 2016-12-17:
Thanks. I understand that there are some libraries for using cc1310/cc1350 LaunchPads, but I would like to connect them to my MSP40 LaunchPad and work with cc1310/cc1350 as a network processor over UART. Is it possible in Energia as well?

rei-vilo, 2016-12-17:
I don't see the benefit. The code for the MSP430 can run on the CC1310 and CC1310 LaunchPads. 

Otherwise, consider the Anaren CC110L BoosterPack.

github4f, 2016-12-18:
Really??? the code for the MSP430 can also run on CC3200 LaunchPads! so why
you used your mind to prepare a Cc3200 library for MSP430?

CC110L is expensive (12-15$ for 500 pcs). If I order it to china, I can
make a cc1310 board for less than 6$

rei-vilo, 2016-12-18:
The Energia framework allows the same code to run on different MCUs. 

Please consider the 43oh.com and e2e.ti.com forums for other answers to your question.

robertinant, 2016-12-20:
There is a CC3100 Library for the MSP430. I do not currently have plans to support the CC1310 as a network processor like we have for the CC110L BoosterPack.