energia/Energia #951: Energia's Wire library doesn't work in MSP432

danielrikmed93, 2016-12-19:
Hello,

I'm have a trouble with the Wire library in the Energia 18, the IDE doesn't recognize any commands, the IDE upload the code to the MSP but when I sense the output with the logic analyzer there is not output.

[image]

rei-vilo, 2016-12-20:
Have a look at documents [msp432p401r](http://www.ti.com/lit/gpn/msp432p401r) and [slac698f](http://www.ti.com/general/docs/lit/getliterature.tsp?baseLiteratureNumber=SLAC698&fileType=zip) but you would need to use CCS.

robertinant, 2016-12-20:
Currently there is only on Wire available. There is no plan to add additional Wire instances at this point.

closed 2016-12-20 by robertinant