energia/Energia #953: microscend interrupt Using OneMsTaskTimer

amirezz, 2016-12-21:
Hi all "
hope you are doing well.
thanks for that great job .
i have project that want to call function every 25 microsecond .i was using timerone library in arduino and it's working fine.
 Now i am using the library OneMsTaskTimer .But i found that it have the scale of millisecond only ,can i edit in that library to fit my needs .
thanks in advance.

robertinant, 2016-12-23:
You can, you will see the timer loaded in OneMsTaskTimer.cpp with either load or PWM_PERIOD depending on the MSP430. In both cases this is F_CPU/1000 which gives you a resolution of 1ms. You can change that load to F_CPU/1000000L which will give you a microsecond resolution. Be careful with this though since the timer interrupt will now fire every microsecond.

amirezz, 2016-12-26:
Hi robertinant "
Thanks for your interest and replay .

I try that now . but it didn't enter the interrupt  function as i expected.
the code is
 OneMsTaskTimer_t myTask1 ={25,read_sensor, 0, 0};

and in setup

  	              DCOCTL = 0x00;  // Set DCOCLK to 1MHz
	              BCSCTL1 = CALBC1_1MHZ;
	              DCOCTL = CALDCO_1MHZ;

 OneMsTaskTimer::start();
  OneMsTaskTimer::add(&myTask1); // 25 us period

FYI it's working fine in arduino in the same frequency(1 mhz) and the same period of interrupt   (25 us).
There is any thing wrong? 
Thanks in advance .