energia/Energia #955: Upload Error to MSP430G2553 launchpad with Energia 18!

abhaysbharadwaj, 2017-01-14:
* LaunchPad board name and version: MSP-EXP430G2 Rev 1.5
* Energia IDE version (found in Energia > About Energia menu): 1.6.10E18
* Board package version (found in Tools > Boards > Boards Manager menu): `INSERT BOARD PACKAGE VERSION`
* OS name and version: Windows 7 x32, Windows 8.1 x86

Error: 
usbutil: unable to find a device matching 0451:f432

I am able to upload the same code using older versions of energia. But in 1.6.10E18, I always get this error! Kindly help.
I have tried uploading different sketches like blink, fade, etc. I face the same issue with the new version ov energia.
[image]

abecedarian01, 2017-07-04:
Same error here- Dell laptop with Windows 10.

Error messages:
________________________________________________________________________
Energia: 1.6.10E18 (Windows 10), Board: "MSP-EXP430G2553LP"

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

C:\energia\18\hardware\energia\msp430\cores\msp430\atof.c: In function 'atof':

C:\energia\18\hardware\energia\msp430\cores\msp430\atof.c:71:9: warning: floating constant exceeds range of 'double' [-Woverflow]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

C:\energia\18\hardware\energia\msp430\cores\msp430\avr\dtostrf.c: In function 'dtostrf':

C:\energia\18\hardware\energia\msp430\cores\msp430\avr\dtostrf.c:36:3: warning: incompatible implicit declaration of built-in function 'sprintf' [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

C:\energia\18\hardware\energia\msp430\cores\msp430\Stream.cpp: In member function 'bool Stream::find(char*)':

C:\energia\18\hardware\energia\msp430\cores\msp430\Stream.cpp:78:30: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

Sketch uses 394 bytes (2%) of program storage space. Maximum is 16,384 bytes.
Global variables use 20 bytes (3%) of dynamic memory, leaving 492 bytes for local variables. Maximum is 512 bytes.
MSPDebug version 0.22 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2013 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
usbutil: unable to find a device matching 0451:f432

An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
________________________________________________________________________

Midun-S, 2018-01-26:
Follow These
1. Copy C:\TIMSP430\energia1.6.10E18-windows\energia-
    1.6.10E18\hardware\tools\DSLite\DebugServer\drivers\MSP430.dll
 
                             to
2. C:\TIMSP430\energia1.6.10E18-windows\energia-
    1.6.10E18\hardware\tools\msp430\bin

3. Goto C:\TIMSP430\energia1.6.10E18-windows\energia-
    1.6.10E18\hardware\energia\msp430\boards.txt
   
       then replace 2 occurrences of rf2500 with tilib
4. Restart Energia you will be able t upload to your board.

Note: use a folder with a name without spaces. 
NOTE: Have the folder in C:\ drive