energia/Energia #956: Board keeps resetting whenever it enters into loop()

abhaysbharadwaj, 2017-01-14:
* LaunchPad board name and version: `MSPEXP430-G2 Rev1.5` MSP430g2553
* Energia IDE version (found in Energia > About Energia menu): `Energia 1.6.10E18`
* OS name and version: `Windows 7 (x32) and Windows 8.1 (x64)`

I have attached my code below.

This code works perfectly with my Arduino Uno R3 board. But When I try it with my MSP430G2553 board (with relevant modifications in code), it starts to execute normally but as soon as it enters the loop(), the board resets and starts allover again.

I have tried it with previous versions of energia as well as the latest version of CCS, Its the same result everytime.

It like, as soon as the code execution enters the loop(), it resets and starts all over again. (I have put print statements at the start of the loop() to verify where it is happening).

Can anyone help me out?

`#include <gsmMqtt.h>
#include <SoftwareSerial.h>
#define DEBUG true
#define callPin P1_3
SoftwareSerial sim808(P2_0, P2_1);

//bolean flags
boolean gprsReady = false;
boolean mqttSent = false;

void setup()
{
  pinMode(RED_LED, OUTPUT);
  pinMode(GREEN_LED, OUTPUT);
  digitalWrite(RED_LED, LOW);
  digitalWrite(GREEN_LED, LOW);
  pinMode(callPin, INPUT);
  digitalWrite(callPin, HIGH);
  Serial.begin(9600);
  sim808.begin(9600);
  Serial.print(("GPS/GSM/GPRS Initializing"));
  Serial.println((""));
  digitalWrite(RED_LED, HIGH);
  boolean ty = isGPRSReady();
  while (ty != true)
  {
      Serial.println(ty);
      Serial.println(("GSM Module not responding!"));
      delay(500);
      boolean ty = isGPRSReady();
  }
  //call();//do a simple call to get AGPS fix.
  delay(5000);
  hangup();
  digitalWrite(RED_LED, LOW);
  Serial.println(("setup ended"));
}

void loop()
{
  digitalWrite(GREEN_LED,HIGH);
  Serial.println(("loop entered"));
  char type[10], no1[3], no2[3];
  char date[20], lat[12], lon[12];
  char buf[150];
  //sos();//make a call if button pressed.
  getgps();//Get GPS data
  String x = sendData( "AT+CGNSINF", 1000, DEBUG);
  //+CGNSINF: 1,1,20161122182451.000,13.019292,77.686463,919.200,0.15,10.5,1,,0.9,2.0,1.8,,12,9,,,47,,
  x.toCharArray(buf, 150);
  strcpy(type, strtok(buf , ":"));
  strcpy(no1, strtok(NULL, ","));
  strcpy(no2 , strtok(NULL, ","));
  strcpy(date , strtok(NULL, ","));
  strcpy(lat, strtok(NULL, ","));
  strcpy(lon, strtok(NULL, ","));

  Serial.print(("Date/Time: ")); Serial.println(date);
  Serial.print(("Latitude : ")); Serial.println(lat);
  Serial.print(("Longitude: ")); Serial.println(lon);
  Serial.println((""));
  Serial.println(("Checking if GPRS is READy?"));
  sim808.println("AT");
  delay(1000);
  gprsReady = isGPRSReady();
  if (gprsReady == false)
  {
    Serial.println(("GPRS Not Ready"));
  }
  if (gprsReady == true)
  {
    digitalWrite(GREEN_LED, HIGH);
    Serial.println(("GPRS READY!"));
    String data = "{";
  data += "\n";
  data += "\"ID\":\"FlipGNS1\",";
  data += "\n";
  data += "\"Date\":\"";
  data += date;
  data += "\",\n";
  data += "\"Latitude\":\"";
  data += lat;
  data += "\",\n";
  data += "\"Longitude\":\"";
  data += lon;
  data += "\"\n";
  data += "}";
    String json = data;
    char jsonStr[200];
    json.toCharArray(jsonStr, 200);
    Serial.println(json);
    //*arguments in function are:
    //clientID, IP, Port, Topic, Message
    sendMQTTMessage("flipgsm1", "52.204.21.160", "1883", "flip/gsm/gps", jsonStr);
    //digitalWrite(GREEN_LED, LOW);
  }
  Serial.println(("loop ended"));
  Serial.println("");
}

void call(void)
{
  sendData( "AT+CSQ", 1000, DEBUG);//check signal quality
  Serial.println(("Making a call!"));
  sim808.println("ATD9663984729;"); // xxxxxxxxx is the number you want to dial.
  if (Serial.available())
  Serial.print((unsigned char)sim808.read());
  delay(10000);
  delay(10000);
}

void hangup(void)
{
    sim808.println("ATH"); //End the call.
    if (sim808.available())
      Serial.print((unsigned char)sim808.read());
    Serial.println(("Call ended!"));
}
void getgps(void)
{
    Serial.println(("Getting GPS..."));
  sendData( "AT+CGNSPWR=1", 1000, DEBUG);//power ON GPS/GNS module
  String x = sendData( "AT+CGNSSTATUS?", 1000, DEBUG);//Check Status
  //wait till 3D or 2D fix is obtained.
  if (x != "Location 3D Fix")
  {
    String x = sendData( "AT+CGPSSTATUS?", 1000, DEBUG);
    Serial.println(x);
  }
  sendData( "AT+CGNSSEQ=RMC", 1000, DEBUG);//read GPRMC data
}

String sendData(String command, const int timeout, boolean debug)
{
  String response = "";
  Serial.println(command);
  long int time = millis();
  while ( (time + timeout) > millis())
  {

    while (Serial.available())
    {
      char c = Serial.read();

      response += c;
    }
  }
  if (debug)
  {
    Serial.print(response);
  }
  return response;
}

boolean isGPRSReady()
{
  int index = 0;
  byte data1;
  String SerialStr = "";
  sim808.println("AT");
  sim808.println("AT");
  sim808.println("AT+CGATT?");
  index = 0;
  while (sim808.available())
  {
    data1 = (char)sim808.read();
    Serial.write(data1);
    SerialStr[index++] = data1;
  }
  Serial.println("Check OK");
  Serial.print("gprs str = ");
  Serial.println(data1);
  if (data1 > -1 && ( data1 !=0 || data1!=57))
  {
    Serial.println("GPRS OK");
    return true;
  }
  else
  {
    Serial.println("GPRS NOT OK");
    return false;
  }
}

void sendMQTTMessage(char* clientId, char* brokerUrl, char* brokerPort, char* topic, char* message)
{
  byte mqttMessage[127];
  int mqttMessageLength = 0;
  char atCommand[50];
  sim808.println("AT"); // Sends AT command to wake up cell phone
  Serial.println("send AT to wake up sim808");
  delay(1000); // Wait a second
  // digitalWrite(13, HIGH);
  sim808.println("AT+CSTT=\"www\",\"\",\"\""); // Puts phone into sim808 mode
  Serial.println("AT+CSTT=\"www\",\"\",\"\"");
  delay(2000); // Wait a second
  sim808.println("AT+CIICR");
  Serial.println("AT+CIICR");
  delay(2000);
  sim808.println("AT+CIFSR");
  Serial.println("AT+CIFSR");
  delay(1000);
  strcpy(atCommand, "AT+CIPSTART=\"TCP\",\"");
  strcat(atCommand, brokerUrl);
  strcat(atCommand, "\",\"");
  strcat(atCommand, brokerPort);
  strcat(atCommand, "\"");
  sim808.println(atCommand);
  Serial.println(atCommand);
  Serial.println("AT+CIPSTART=\"TCP\",\"mqttdashboard.com\",\"1883\"");
  delay(1000);
  sim808.println("AT+CIPSEND");
  Serial.println("AT+CIPSEND");
  delay(1000);
  mqttMessageLength = 16 + strlen(clientId);
  Serial.println(mqttMessageLength);
  mqtt_connect_message(mqttMessage, clientId);
  for (int j = 0; j < mqttMessageLength; j++)
  {
    sim808.write(mqttMessage[j]); // Message contents
    Serial.write(mqttMessage[j]); // Message contents
    Serial.println("");
  }
  sim808.write(byte(26)); // (signals end of message)
  Serial.println("Sending message");
  delay(1000);
  sim808.println("AT+CIPSEND");
  Serial.println("AT+CIPSEND");
  delay(1000);
  mqttMessageLength = 4 + strlen(topic) + strlen(message);
  Serial.println(mqttMessageLength);
  mqtt_publish_message(mqttMessage, topic, message);
  for (int k = 0; k < mqttMessageLength; k++)
  {
    sim808.write(mqttMessage[k]);
    Serial.write((byte)mqttMessage[k]);
  }
  sim808.write(byte(26)); // (signals end of message)
  Serial.println("");
  Serial.println("-------------Sent-------------"); // Message contents
  delay(2000);
  sim808.println("AT+CIPCLOSE");
  Serial.println("AT+CIPCLOSE");
  delay(1000);
}

void sos()
{
  int xx = digitalRead(callPin);
  if (xx == 0)
  {
    Serial.println("SOS Activated");
    call();
    delay(20000);
    Serial.println("SOS Ended");
  }
  xx == 1;
  hangup();
}
`

P.S. because the dynamic memory of my code reaches 75%, I have reduced the Software Serial RX buffer to 10 (from 16) in the SoftwareSerial.h file. (This shouldn't matter I presume)

[sim808_mqtt_energia1.txt](https://github.com/energia/Energia/files/706224/sim808_mqtt_energia1.txt)

rei-vilo, 2017-01-14:
It looks like you're facing a RAM overflow. Keep in mind the MSP430G2553 has only 512 bytes of RAM. 

For project-related questions, please consider the http://43oh.com instead.

closed 2017-01-14 by rei-vilo

abhaysbharadwaj, 2017-01-14:
Hello,
Thank you for the 43oh reference. I will do that.

Regarding RAM overflow, I am well within the limits. Refer the attached pic:

[image]

rei-vilo, 2017-01-14:
Yes, the figure is correct for declared variable but the SIM808 library may create variable dynamically.