energia/Energia #958: macOS — Empty /ti Sub-Folder Created Under HOME Folder

rei-vilo, 2017-02-23:
When launching a build though **Verify/Compile** or **Upload**, Energia creates a sub-folder under the home folder, in my case `/Users/ReiVilo/ti` on macOS.

The location of the sub-folder under the `home` folder is not the most appropriate. Furthermore, it is empty!

robertinant, 2017-02-23:
I am able to reproduce this when hitting upload. It is DSLite that is creating the directory. I will reach out to the DSLite team to ask why they are creating that directory and what it's purpose is.

robertinant, 2017-03-01:
The DSLite team informed me that this is a know issue and will be fixed in a future version. Unfortunately it won't make it into the March release but should go into the one after that.