energia/Energia #960: expected primary-expression before ',' token

oTijWi, 2017-02-28:
here is my code -

`#define LED
void setup()
{                
  pinMode( LED , OUTPUT);     
}
void loop()
{
  digitalWrite( LED , HIGH);   // turn the LED on by making the LED pin HIGH
  delay(100000);               // wait for a second
  digitalWrite( LED , LOW);    // turn the LED off by making the pin LOW
  delay(100000);               // wait for a second
}`
 
& ERRORS -> 

Energia: 1.6.10E18 (Windows 10), Board: "MSP-EXP430G2553LP"

<command-line>:0:12: warning: missing whitespace after the macro name [enabled by default]

C:\Users\user\Desktop\desk\Energia\Sidekick_LaunchPad\EXAMPLE00_Blink\EXAMPLE00_Blink.ino: In function 'void setup()':

EXAMPLE00_Blink:4: error: expected primary-expression before ',' token

C:\Users\user\Desktop\desk\Energia\Sidekick_LaunchPad\EXAMPLE00_Blink\EXAMPLE00_Blink.ino: In function 'void loop()':

EXAMPLE00_Blink:8: error: expected primary-expression before ',' token

EXAMPLE00_Blink:10: error: expected primary-expression before ',' token

exit status 1
expected primary-expression before ',' token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

rei-vilo, 2017-02-28:
Take the `Blink` example with the full line `#define LED RED_LED` including **RED_LED**.

Please use the [43oh forum](http://www.43oh.com) for project related questions.

```
#define LED RED_LED

void setup() {                
  pinMode(LED, OUTPUT);     
}

void loop() {
  digitalWrite(LED, HIGH);
  delay(1000);              
  digitalWrite(LED, LOW);
  delay(1000);              
}
```

closed 2017-02-28 by rei-vilo

oTijWi, 2017-03-02:
[image]

now, this problem!!! 

C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\tools\msp430/bin/msp430-ar: unable to rename 'core\core.a'; reason: Permission denied

exit status 1
Error compiling for board MSP-EXP430G2553LP.

oTijWi, 2017-03-02:
Please...
help as soon as possible.

rei-vilo, 2017-03-02:
Isn't the desktop write-protected in Windows?

robertinant, 2017-03-03:
I have not seen this problem before. Can you please try the following?

- Close Energia
- Open Energia
- Go to file->Preferences and check "compilation" next to "Show verbose output during:"
- Click the verify button
- If the problem still exists, copy past all text from the black window into this issue.

oTijWi, 2017-03-03:
C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\arduino-builder -dump-prefs -logger=machine -hardware "C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware" -tools "C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\tools-builder" -tools "C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\tools\msp430" -built-in-libraries "C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\libraries" -libraries "C:\Users\user\Documents\Energia\libraries" -fqbn=energia:msp430:MSP-EXP430G2553LP -ide-version=10610 -build-path "C:\Users\user\AppData\Local\Temp\build530c262522cbfcd5f8692c0ec053a60a.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\user\Desktop\desk\Energia\Sidekick_LaunchPad\Sidekick_LaunchPad\EXAMPLE00_Blink\EXAMPLE00_Blink.ino"
C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\arduino-builder -compile -logger=machine -hardware "C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware" -tools "C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\tools-builder" -tools "C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\tools\msp430" -built-in-libraries "C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\libraries" -libraries "C:\Users\user\Documents\Energia\libraries" -fqbn=energia:msp430:MSP-EXP430G2553LP -ide-version=10610 -build-path "C:\Users\user\AppData\Local\Temp\build530c262522cbfcd5f8692c0ec053a60a.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\user\Desktop\desk\Energia\Sidekick_LaunchPad\Sidekick_LaunchPad\EXAMPLE00_Blink\EXAMPLE00_Blink.ino"
Using board 'MSP-EXP430G2553LP' from platform in folder: C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\energia\msp430
Using core 'msp430' from platform in folder: C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\energia\msp430
Detecting libraries used...
"C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\tools\msp430/bin/msp430-g++" -c -g -O2  -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=msp430g2553 -DF_CPU=16000000L -DARDUINO=10610 -DENERGIA=10610 -DENERGIA_MSP-EXP430G2553LP -DENERGIA_ARCH_MSP430 -IC:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\tools\msp430/include  "-IC:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\energia\msp430\cores\msp430" "-IC:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\energia\msp430\variants\MSP-EXP430G2553LP" "C:\Users\user\AppData\Local\Temp\build530c262522cbfcd5f8692c0ec053a60a.tmp\sketch\EXAMPLE00_Blink.ino.cpp" -o "nul"
"C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\tools\msp430/bin/msp430-g++" -c -g -O2  -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=msp430g2553 -DF_CPU=16000000L -DARDUINO=10610 -DENERGIA=10610 -DENERGIA_MSP-EXP430G2553LP -DENERGIA_ARCH_MSP430 -IC:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\tools\msp430/include  "-IC:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\energia\msp430\cores\msp430" "-IC:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\energia\msp430\variants\MSP-EXP430G2553LP" "C:\Users\user\AppData\Local\Temp\build530c262522cbfcd5f8692c0ec053a60a.tmp\sketch\EXAMPLE00_Blink.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\tools\msp430/bin/msp430-g++" -c -g -O2  -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=msp430g2553 -DF_CPU=16000000L -DARDUINO=10610 -DENERGIA=10610 -DENERGIA_MSP-EXP430G2553LP -DENERGIA_ARCH_MSP430 -IC:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\tools\msp430/include  "-IC:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\energia\msp430\cores\msp430" "-IC:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\energia\msp430\variants\MSP-EXP430G2553LP" "C:\Users\user\AppData\Local\Temp\build530c262522cbfcd5f8692c0ec053a60a.tmp\sketch\EXAMPLE00_Blink.ino.cpp" -o "C:\Users\user\AppData\Local\Temp\build530c262522cbfcd5f8692c0ec053a60a.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\tools-builder\ctags\5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\user\AppData\Local\Temp\build530c262522cbfcd5f8692c0ec053a60a.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\tools\msp430/bin/msp430-g++" -c -g -O2  -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=msp430g2553 -DF_CPU=16000000L -DARDUINO=10610 -DENERGIA=10610 -DENERGIA_MSP-EXP430G2553LP -DENERGIA_ARCH_MSP430 -IC:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\tools\msp430/include  "-IC:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\energia\msp430\cores\msp430" "-IC:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\energia\msp430\variants\MSP-EXP430G2553LP" "C:\Users\user\AppData\Local\Temp\build5 [... cut ...]
"C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\tools\msp430/bin/msp430-ar" rcs  "C:\Users\user\AppData\Local\Temp\build530c262522cbfcd5f8692c0ec053a60a.tmp\core\core.a" "C:\Users\user\AppData\Local\Temp\build530c262522cbfcd5f8692c0ec053a60a.tmp\core\twi.c.o"
C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\tools\msp430/bin/msp430-ar: unable to rename 'C:\Users\user\AppData\Local\Temp\build530c262522cbfcd5f8692c0ec053a60a.tmp\core\core.a'; reason: File exists

exit status 1
Error compiling for board MSP-EXP430G2553LP.

oTijWi, 2017-03-03:
again getting a problem,
but this time,
reason is changed.

rei-vilo, 2017-03-03:
This seems to be a second compilation, not a fresh one. 

Please close Energia, relaunch it, create a new project with `Blink` example, compile.

oTijWi, 2017-03-03:
hey, rei-vilo. I didn't get what you were saying.
will you be more specific?

oTijWi, 2017-03-03:
same problem again :--

C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\tools\msp430/bin/msp430-ar: unable to rename 'C:\Users\user\AppData\Local\Temp\build6bdd3e6f9b1f6496b4adec1c9457df9c.tmp\core\core.a'; reason: Permission denied

exit status 1
Error compiling for board MSP-EXP430G2553LP.

rei-vilo, 2017-03-03:
There seems to be an issue with folder permissions. 

- Have you installed Energia with administrator privilege?
- Which OS and release are you using?

oTijWi, 2017-03-03:
I'm using windows 10.
about the installation of energia, I didn't do it.
actually, I've got a zip file for it which was having energia preinstalled.

rei-vilo, 2017-03-03:
Perform a check of your disk and repair files permissions.

oTijWi, 2017-03-03:
can you guide me, how to do that?

rei-vilo, 2017-03-03:
GIYF

https://support.microsoft.com/en-us/help/17590/automatically-diagnose-and-repair-windows-file-and-folder-problems

oTijWi, 2017-03-03:
Again having the same problem

C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\tools\msp430/bin/msp430-ar: unable to rename 'C:\Users\user\AppData\Local\Temp\build31e818c65d9c7a19f0d6d43124a20cd2.tmp\core\core.a'; reason: Permission denied

exit status 1
Error compiling for board MSP-EXP430G2553LP.

oTijWi, 2017-03-03:
Maybe it is because I haven't installed Energia manually,
But, I couldn't get to download the set up for it.
it is available in ZIP form to download.

rei-vilo, 2017-03-03:
So this is an issue with Windows, not Energia per se.

Ok, now try to launch Energia with Administrator privilege.

oTijWi, 2017-03-03:
again the same problem ..... 
:(

C:\Users\user\Desktop\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\tools\msp430/bin/msp430-ar: unable to rename 'C:\Users\user\AppData\Local\Temp\build99cc971bd16d54daa628c37bf2ac45d8.tmp\core\core.a'; reason: File exists

exit status 1
Error compiling for board MSP-EXP430G2553LP.

oTijWi, 2017-03-03:
this time, it isn't saying about the permission.
but the reason is changed.
 now the error is because of the existence of a file.
Don't know what to do now.

rei-vilo, 2017-03-03:
Delete the file.

rei-vilo, 2017-03-03:
Maybe you'd be luckier if you post your issue on the [43oh forum](http://43oh.com). Other users may have cope with this issue before. I can't manage to replicate it on my Windows 10-based laptop.

oTijWi, 2017-03-03:
I've been trying to get this issue on 43oh, but there is a question I need to answer to get registered.
What is the name of TI's flagship development kit for makers?

when I'm filling the name of the msp430 launchpad it is again asking me for a correct answer.
so, couldn't post it on that forum unless I register on that.

rei-vilo, 2017-03-03:
I guess it is `launchpad`.

rei-vilo, 2017-03-03:
And what about 

> Delete the file `C:\Users\user\AppData\Local\Temp\build99cc971bd16d54daa628c37bf2ac45d8.tmp\core\core.a` or the folder `C:\Users\user\AppData\Local\Temp\build99cc971bd16d54daa628c37bf2ac45d8.tmp`.