energia/Energia #963: using Wire.setModule()  with a value greater than 3 will "soft-brick" the board

jstamour802, 2017-03-09:
* LaunchPad board name and version: `EK-TM4C1294XL`
* Energia IDE version (found in Energia > About Energia menu): `1.6.10E18`
* Board package version (found in Tools > Boards > Boards Manager menu): `1.0.2`
* OS name and version: `Windows 10 Pro`

I am expecting the ability to use all of the i2c modules on the TM4C1294, but when I enter a Wire.setModule(n) greater than 3 the board becomes unresponsive. I can then no longer program any other sketches onto the board until I run the LM Flash Tool debug port unlock.

Wire.setModule(0) thru 3 do not cause this problem.

steps to reproduce the problem:

  Setup a program to blink an led.  Use wire.setModule(0) & wire.Begin and the program will run as expected. Change wire.setModule to 4 or higher to break it.

note: you'll need lm flash tool and run the debug port unlock in order to program the launchpad again after breaking it.

rei-vilo, 2017-03-09:
There are 4 I²C ports on the TM4C, numbered `0` through `3`. 

So the valid values for `Wire.setModule()` are `0`, `1`, `2`, and `3`.

closed 2017-03-09 by rei-vilo

jstamour802, 2017-03-09:
the TM4C1294 has 10 i2c ports. So it's not possible to address the 6 other i2c ports available?

rei-vilo, 2017-03-09:
Energia implements 4 ports.

robertinant, 2017-03-09:
As @rei-vilo mentioned we currently only support 4 I2C ports and none of the other ports are currently accessible. With that said, it should not crash / brick when using a number higher than 3 in setModule. I'll look into this to try and figure out why it is crashing / bricking the chip.