energia/Energia #964: SoftwareSerial For cc13xx

denzelwamburu, 2017-03-21:
Thank you for opening an issue on the Energia repository. 

### Do not use GitHub issues for project related questions. 

Use instead the [43oh forum](http://forum.43oh.com) and [TI E2E forum](http://e2e.ti.com), so everyone can share the questions and answers.

* LaunchPad board name and version: `cc1310 Rev 1.1`
* Energia IDE version (found in Energia > About Energia menu): `1.6.10E18`
* Board package version (found in Tools > Boards > Boards Manager menu): `cc1310 EMT`
* OS name and version: `UBUNTU 16`

### Describe the expected results.

`Am trying to do a Software Serial on cc1310 Board. The current core cc13xx library seems not to support it. I've tried to port the msp430 lib but no results. Any workaround would be appreciated `

denzelwamburu, 2017-03-21:
Porting the Arduino library throws the error below

`/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:260:2: error: #error This version of SoftwareSerial supports only 20, 16 and 8MHz processors
 #error This version of SoftwareSerial supports only 20, 16 and 8MHz processors
  ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp: In member function 'bool SoftwareSerial::listen()':
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:316:23: error: 'SREG' was not declared in this scope
     uint8_t oldSREG = SREG;
                       ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:317:9: error: 'cli' was not declared in this scope
     cli();
         ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp: In member function 'void SoftwareSerial::setTX(uint8_t)':
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:491:44: error: 'digitalPinToBitMask' was not declared in this scope
   _transmitBitMask = digitalPinToBitMask(tx);
                                            ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:492:37: error: 'digitalPinToPort' was not declared in this scope
   uint8_t port = digitalPinToPort(tx);
                                     ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:493:50: error: 'portOutputRegister' was not declared in this scope
   _transmitPortRegister = portOutputRegister(port);
                                                  ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp: In member function 'void SoftwareSerial::setRX(uint8_t)':
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:502:43: error: 'digitalPinToBitMask' was not declared in this scope
   _receiveBitMask = digitalPinToBitMask(rx);
                                           ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:503:37: error: 'digitalPinToPort' was not declared in this scope
   uint8_t port = digitalPinToPort(rx);
                                     ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:504:48: error: 'portInputRegister' was not declared in this scope
   _receivePortRegister = portInputRegister(port);
                                                ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp: In member function 'void SoftwareSerial::begin(long int)':
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:515:31: error: 'table' was not declared in this scope
   for (unsigned i=0; i<sizeof(table)/sizeof(table[0]); ++i)
                               ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:531:38: error: 'digitalPinToPCICR' was not declared in this scope
     if (digitalPinToPCICR(_receivePin))
                                      ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:533:78: error: 'digitalPinToPCICRbit' was not declared in this scope
       *digitalPinToPCICR(_receivePin) |= _BV(digitalPinToPCICRbit(_receivePin));
                                                                              ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:533:79: error: '_BV' was not declared in this scope
       *digitalPinToPCICR(_receivePin) |= _BV(digitalPinToPCICRbit(_receivePin));
                                                                               ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:534:37: error: 'digitalPinToPCMSK' was not declared in this scope
       *digitalPinToPCMSK(_receivePin) |= _BV(digitalPinToPCMSKbit(_receivePin));
                                     ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:534:78: error: 'digitalPinToPCMSKbit' was not declared in this scope
       *digitalPinToPCMSK(_receivePin) |= _BV(digitalPinToPCMSKbit(_receivePin));
                                                                              ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp: In member function 'void SoftwareSerial::end()':
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:549:36: error: 'digitalPinToPCMSK' was not declared in this scope
   if (digitalPinToPCMSK(_receivePin))
                                    ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:550:77: error: 'digitalPinToPCMSKbit' was not declared in this scope
     *digitalPinToPCMSK(_receivePin) &= ~_BV(digitalPinToPCMSKbit(_receivePin));
                                                                             ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:550:78: error: '_BV' was not declared in this scope
     *digitalPinToPCMSK(_receivePin) &= ~_BV(digitalPinToPCMSKbit(_receivePin));
                                                                              ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp: In member function 'virtual size_t SoftwareSerial::write(uint8_t)':
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:585:21: error: 'SREG' was not declared in this scope
   uint8_t oldSREG = SREG;
                     ^
[... cut ...]
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp: In member function 'virtual void SoftwareSerial::flush()':
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:633:21: error: 'SREG' was not declared in this scope
   uint8_t oldSREG = SREG;
                     ^
/home/wamburu/Energia/libraries/SoftwareSerial-master/SoftwareSerial.cpp:634:7: error: 'cli' was not declared in this scope
   cli();
       ^
exit status 1
Error compiling for board LaunchPad w/ cc1310 EMT (48MHz).`

rei-vilo, 2017-03-21:
Please post your question to the [43oh forum](http://forum.43oh.com).

About libraries:

* [How To: Porting Libraries](http://forum.43oh.com/topic/7511-howto-porting-libraries-some-help-needed/)
* [Some Misconceptions about Libraries](http://forum.43oh.com/topic/6944-posting-a-library-for-energia/)
* [Posting a Library for Energia](http://forum.43oh.com/topic/6944-posting-a-library-for-energia/)

closed 2017-03-21 by rei-vilo