energia/Energia #965: When are we going delete this repository?

RickKimball, 2017-03-21:
This github repository is no longer the source code for the latest Energia. When are you going to either merge the real source code in or delete this repo?

rei-vilo, 2017-03-21:
Should this very repository be deleted, what would happen to all the issues (still un-answered) and pull-requests (not committed)?

RickKimball, 2017-03-21:
Obviously, I don't want to delete this repo. What I'd like is for Energia development to take place like normal open source projects take place. This is an attempt to wake people up to the fact that this repository isn't the real stuff.  I don't understand why you are doing commits on this tree when it doesn't relate to the current tree

It doesn't seem too much to ask for this minimal goals:
- public development of the source code
- a preference for open source code instead of the binary blobs
- release tags in the repo so i can rebuild from source for a revision.
- release candidates that people can try before completely untested releases are just burped out

RickKimball, 2017-03-21:
> what would happen to all the issues (still un-answered) and pull-requests (not committed)?
There are pull-requests in there for over 3 years that are ready to go and have never been acted on. What makes you think someone is ever going to actually merge/test and commit them?

bombledmonk, 2017-03-30:
Excuse my ignorance, but where is the official source code located?

robertinant, 2017-03-31:
The IDE sources are here: https://github.com/robertinant/EnergiaNG. This includes the current MSP430 support. The tag is 1.6.10E18 for the current release of the IDE. Since then we have moved MSP430 into it's own repository decoupling it from the IDE. The MSP430 repo that will be included in the next MSP430 package update / IDE update is located here: https://github.com/energia/msp430-lg-core

Hope this helps.

bombledmonk, 2017-04-04:
Thanks.  I'm sure it's always hard tie up loose ends or you may not be ready right now, but might it make sense to make note on this repo's main Readme.md and possibly the main website?  This still referenced as the main repo on the web.

cambazz, 2017-04-21:
If this is not the real source tree, where will we submit the issues?

rei-vilo, 2017-04-21:
Depending on the issue you want to report, 

* If the issue is about a specific MCU, look for the [corresponding repository](https://github.com/energia) ending with `-core`.

> Example: use [cc13xx-core](https://github.com/energia/cc13xx-core) for the LaunchPad CC1310 and LaunchPad CC1350.

* For general issues about Energia, you can use this very repository, or go to [robertinant/EnergiaNG](https://github.com/robertinant/EnergiaNG).