energia/Energia #971: MSP 432 USB port not showing on OS X

zappfinger, 2017-04-20:
On OS X 10.10.3 I installed 1.6 E1.18. It does not show my connected MSP-EXP432P402R in the USB ports

rei-vilo, 2017-04-20:
Have you installed the MSP432 package with the *Boards Manager*?

zappfinger, 2017-04-20:
Yes I did. But i cannot upload a sketch, it gives an error. I have to be 
able to select a port first..

rei-vilo, 2017-04-20:
There is no automatic port selection on Energia.

zappfinger, 2017-04-20:
I am not suggesting there is. On all Arduino like editors I have to 
select the port first and then the board. This is also described in the 
energia Guide for OS X:

    Selecting the Serial Port

Select/*Serial Port*/from the/*Tools*/menu to view the available serial 
ports. On windows, they will be listed as COMXXX port and usually a 
higher number is the LaunchPad com port. On Mac OS X they will be listed 
as /dev/cu.uart-XXXX./*On Mac OS X make sure to select the 
/dev/tty.uart-XXXX*/

Except for me, there is no /dev/tty.uart-XXXX or similar present

rei-vilo, 2017-04-20:
Power-cycle the Mac, then open a Terminal window, and launch

```
ls /dev/tty.*
```

zappfinger, 2017-04-21:
It was the USB cable. It worked with another USB cable. Sorry for the 
fuss...

robertinant, 2017-04-21:
Thanks for reporting back!

closed 2017-04-21 by robertinant

Kegovc, 2022-05-05:
in Mac osx 11.6.5, the system does not find the msp430 board but in windows it does appear.