energia/Energia #972: analogRead examples give error on MSP432

zappfinger, 2017-04-24:
All analogRead examples give this error:
sketch/AnalogInput.ino.cpp.o: In function `loopAnalogInput()':
/private/var/folders/r0/103fxm8x32j76d1yv589d0jc0000gn/T/arduino_modified_sketch_266087/AnalogInput.ino:41: undefined reference to `analogRead'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board RED LaunchPad w/ msp432 EMT (48MHz).

Is analogRead()  suppose to work on the MSP432?
I have also tried to use analogReadResolution first, to no avail.

robertinant, 2017-04-24:
This has been fixed in the latest core release 4.9.1. Please install this core version through the board manager in Energia.

closed 2017-04-24 by robertinant

zappfinger, 2017-04-24:
Thanks! That seems to work....