energia/Energia #973: I am using msp430 launchpad  and i want to change spi module from UCB0 to UCA0. What should i do???

hemangjoshi37a, 2017-05-03:
Thank you for opening an issue on the Energia repository. 

### Do not use GitHub issues for project related questions. 

Use instead the [43oh forum](http://forum.43oh.com) and [TI E2E forum](http://e2e.ti.com), so everyone can share the questions and answers.

### Do not use GitHub issues for questions on third-party libraries. 

Use instead their own repositories.

### Check the issue hasn't been reported yet.

If an issue has alreday been reported, add your case to its thread.

### Please provide the following information.

* LaunchPad board name and version: `INSERT LAUNCHPAD BOARD NAME AND VERSION`
* Energia IDE version (found in Energia > About Energia menu): `INSERT ENERGIA VERSION`
* Board package version (found in Tools > Boards > Boards Manager menu): `INSERT BOARD PACKAGE VERSION`
* OS name and version: `INSERT OS NAME AND VERSION`

### Describe the expected results.

`INSERT DESCRIPTION OF THE EXPECTED RESULT`

### Describe the actual results.

`INSERT DESCRIPTION OF THE ACTUAL RESULT`

### List the steps to reproduce the problem.

`INSERT STEPS TO REPRODUCE THE PROBLEM`

rei-vilo, 2017-05-04:
Which exact MCU are you using? Please use the template provided when opening a new ticket.

Four options:

* Use Code Composer Studio, which provides access to all resources of the MCUs.
* Pay visit to the [43oh forum](http://forum.43oh.com) and [perform a search](http://forum.43oh.com/search/?&q=UCB0%20UCA0&sortby=relevancy)
* Edit the files of Energia corresponding to the SPI library.
* Use software SPI with [shiftOut()](http://energia.nu/reference/shiftout/) and [shiftIn()](http://energia.nu/reference/shiftin/)

closed 2017-05-04 by rei-vilo

hemangjoshi37a, 2017-05-05:
@rei-vilo
Okay bro. Sure....