energia/Energia #980: Energia IDE does not open GUI in fedora 26

JasonSmitR, 2017-07-23:
### Do not use GitHub issues for project related questions. 

After upgrading to fedora 26 the  1.6.10E18 IDE crashes. The only output is

`Picked up JAVA_TOOL_OPTIONS: 
java.lang.Error: Cannot load com.sun.java.swing.plaf.gtk.GTKLookAndFeel
	at javax.swing.UIManager.initializeDefaultLAF(UIManager.java:1351)
	at javax.swing.UIManager.initialize(UIManager.java:1459)
	at javax.swing.UIManager.maybeInitialize(UIManager.java:1426)
	at javax.swing.UIManager.getLookAndFeel(UIManager.java:492)
	at processing.app.linux.GTKLookAndFeelFixer.installGtkPopupBugWorkaround(GTKLookAndFeelFixer.java:79)
	at processing.app.linux.Platform.setLookAndFeel(Platform.java:40)
	at processing.app.Base.guardedMain(Base.java:214)
      at processing.app.Base.main(Base.java:136)`

### Describe the expected results.
The IDE should appear after the logo but it does not.

### Describe the actual results.

`Nothing at all happens the process just exits`

### List the steps to reproduce the problem.

`Run energia on the latest fedora. Not sure if this is related but a similar program occurred with Arduino (Which energia is based upon?) this was fixed with [https://bugzilla.redhat.com/show_bug.cgi?id=1444550](url) `

MatteoRagni, 2017-10-25:
As for now, I solved by modifying the file `$(which energia)`, commenting line 28 (the line that sets `JAVA_OPTIONS`) and by launching with:

```
_JAVA_OPTIONS="" energia
```

JasonSmitR, 2017-11-09:
Thanks @MatteoRagni  that works like a charm

orlandofloresti, 2018-03-16:
Thanks, @MatteoRagni  works for me too.

deepaklorkhatri007, 2018-03-25:
try this technique https://youtu.be/7swyUs6QzPo

FernandoFong, 2018-04-12:
Thank you @MatteoRagni, you were right!

cyphunk, 2018-08-23:
This issue also appears on latest Archlinux:
```
java.lang.Error: Cannot load com.sun.java.swing.plaf.gtk.GTKLookAndFeel
```
To resolve, similar to @MatteoRagni suggestion, in the ``energia`` script  change:
```
JAVA_OPTIONS=("-DAPP_DIR=$APPDIR" "-Dswing.defaultlaf=com.sun.java.swing.plaf.gtk.GTKLookAndFeel")
```
to:
```
JAVA_OPTIONS=("-DAPP_DIR=$APPDIR" )
```