energia/Energia #984: Energia - IDE font size tiny on HD 1080 screens, barely possible to distinguish stuff

KingZuluKing, 2017-09-22:
Thank you for opening an issue on the Energia repository. 

### Do not use GitHub issues for project related questions. 

Use instead the [43oh forum](http://forum.43oh.com) and [TI E2E forum](http://e2e.ti.com), so everyone can share the questions and answers.

### Do not use GitHub issues for questions on third-party libraries. 

Use instead their own repositories.

### Check the issue hasn't been reported yet.

If an issue has alreday been reported, add your case to its thread.

### Please provide the following information.

* LaunchPad board name and version: `INSERT LAUNCHPAD BOARD NAME AND VERSION`
* Energia IDE version (found in Energia > About Energia menu): `INSERT ENERGIA VERSION`
* Board package version (found in Tools > Boards > Boards Manager menu): `INSERT BOARD PACKAGE VERSION`
* OS name and version: `INSERT OS NAME AND VERSION`

### Describe the expected results.

`INSERT DESCRIPTION OF THE EXPECTED RESULT`

### Describe the actual results.

`INSERT DESCRIPTION OF THE ACTUAL RESULT`

### List the steps to reproduce the problem.

`INSERT STEPS TO REPRODUCE THE PROBLEM`

rei-vilo, 2017-09-22:
@KingZuluKing Please fill-in the section **Please provide the following information**.

KingZuluKing, 2017-10-19:
TM4C129 Connected LaunchPad (EK-TM4C1294XL) Launchpad.
Energia Release version = Release 0101E0017 
OS version = Windows 8.1 Pro, 64bit 
-----------------------------------------------------------
The Energia IDE window does not scale on the HD screen which has resolution of HD 1920x1080. The hole text and is very small, up to completely unusable, even if one changes the font size in settings, it does not help. 

Issue is 100% reproducible

rei-vilo, 2017-10-19:
See #968