energia/Energia #990: Energia 0101E0017  can't serial.print(float) on windows 10

ChikamaYan, 2017-11-16:
### Please provide the following information.

* LaunchPad board name and version: `MSP430G2`
* Energia IDE version (found in Energia > About Energia menu): `0101E0017 `
* Board package version (found in Tools > Boards > Boards Manager menu): `no board manager`
* OS name and version: `windows10`

`Following all of my code:
void setup()
{
}
void loop()
{
  float test = 1.0;
  Serial.print(test);
}
However I get a large number of errors when attempting to compile it:

c:/users/chikamayan/downloads/compressed/energia-0101e0017-windows/energia-0101e0017/hardware/tools/msp430/bin/../lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld.exe: sketch_nov16a.cpp.elf section `.text' will not fit in region `rom'
c:/users/chikamayan/downloads/compressed/energia-0101e0017-windows/energia-0101e0017/hardware/tools/msp430/bin/../lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld.exe: section .vectors loaded at [0000ffe0,0000ffff] overlaps section .text loaded at [0000f800,00010b85]
c:/users/chikamayan/downloads/compressed/energia-0101e0017-windows/energia-0101e0017/hardware/tools/msp430/bin/../lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld.exe: region `rom' overflowed by 3264 bytes
c:/users/chikamayan/downloads/compressed/energia-0101e0017-windows/energia-0101e0017/hardware/tools/msp430/bin/../lib/gcc/msp430/4.6.3\libcrt0.a(_copy_data.o): In function `__do_copy_data':
/root/mspgcc-20120406/BUILD/gcc/gcc/../../../gcc-4.6.3/gcc/config/msp430/crt0.S:195: relocation truncated to fit: R_MSP430_16_BYTE against symbol `__data_load_start' defined in *ABS* section in sketch_nov16a.cpp.elf
core.a(TimerSerial.cpp.o): In function `TimerSerial::TimerSerial()':
TimerSerial.cpp:(.text._ZN11TimerSerialC2Ev+0x10): relocation truncated to fit: R_MSP430_16_BYTE against symbol `vtable for TimerSerial' defined in .rodata._ZTV11TimerSerial section in core.a(TimerSerial.cpp.o)
c:/users/chikamayan/downloads/compressed/energia-0101e0017-windows/energia-0101e0017/hardware/tools/msp430/bin/../lib/gcc/msp430/4.6.3/crt0ivtbl16.o: In function `__isr_14':
(.text+0x2): relocation truncated to fit: R_MSP430_16 against symbol `_unexpected_' defined in .text.crt0 section in c:/users/chikamayan/downloads/compressed/energia-0101e0017-windows/energia-0101e0017/hardware/tools/msp430/bin/../lib/gcc/msp430/4.6.3\libcrt0.a(_unexpected_.o)
core.a(Print.cpp.o): In function `Print::printFloat(double, unsigned char)':
Print.cpp:(.text._ZN5Print10printFloatEdh+0x22): relocation truncated to fit: R_MSP430_16 against symbol `__ltsf2' defined in .text section in c:/users/chikamayan/downloads/compressed/energia-0101e0017-windows/energia-0101e0017/hardware/tools/msp430/bin/../lib/gcc/msp430/4.6.3\libgcc.a(_lt_sf.o)
Print.cpp:(.text._ZN5Print10printFloatEdh+0x52): relocation truncated to fit: R_MSP430_16 against symbol `__divsf3' defined in .text section in c:/users/chikamayan/downloads/compressed/energia-0101e0017-windows/energia-0101e0017/hardware/tools/msp430/bin/../lib/gcc/msp430/4.6.3\libgcc.a(_div_sf.o)
Print.cpp:(.text._ZN5Print10printFloatEdh+0x68): relocation truncated to fit: R_MSP430_16 against symbol `__addsf3' defined in .text section in c:/users/chikamayan/downloads/compressed/energia-0101e0017-windows/energia-0101e0017/hardware/tools/msp430/bin/../lib/gcc/msp430/4.6.3\libgcc.a(_addsub_sf.o)
Print.cpp:(.text._ZN5Print10printFloatEdh+0x78): relocation truncated to fit: R_MSP430_16 against symbol `__floatunsisf' defined in .text section in c:/users/chikamayan/downloads/compressed/energia-0101e0017-windows/energia-0101e0017/hardware/tools/msp430/bin/../lib/gcc/msp430/4.6.3\libgcc.a(_usi_to_sf.o)
Print.cpp:(.text._ZN5Print10printFloatEdh+0x84): relocation truncated to fit: R_MSP430_16 against symbol `__subsf3' defined in .text section in c:/users/chikamayan/downloads/compressed/energia-0101e0017-windows/energia-0101e0017/hardware/tools/msp430/bin/../lib/gcc/msp430/4.6.3\libgcc.a(_addsub_sf.o)
Print.cpp:(.text._ZN5Print10printFloatEdh+0xa2): relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
Print.cpp:(.text._ZN5Print10printFloatEdh+0xba): relocation truncated to fit: R_MSP430_16 against symbol `__mulsf3' defined in .text section in c:/users/chikamayan/downloads/compressed/energia-0101e0017-windows/energia-0101e0017/hardware/tools/msp430/bin/../lib/gcc/msp430/4.6.3\libgcc.a(_mul_sf.o)
Print.cpp:(.text._ZN5Print10printFloatEdh+0xc2): additional relocation overflows omitted from the output
collect2: ld returned 1 exit status

Anyone knows what's the cause?
Ps. I've got to use 0101E0017 to compile for MSP430
`

rei-vilo, 2017-11-16:
* Serial isn't initialised.
* Memory overflow. The program is too large for the Flash.
* Please post your issue to the 43oh.com forum.

closed 2017-11-16 by rei-vilo

StefanSch, 2017-11-17:
Did you select a board in the board manager? Looks like no Launchpad is selected.

ChikamaYan, 2017-11-17:
0101E0017 version of Energia doesn't seem to have a board manager....

robertinant, 2017-11-17:
Looks like you compiled for the G2231. This MSP430 has way to little RAM/FLASH to be able to support floats. Either use a larger memory MSP430 or not use floats. I would advice against using floats on micro's that do not have a floating point unit.

ChikamaYan, 2017-11-18:
That perfectly solved my problem! Thank you