energia/Energia #994: MSP430F5529 flash memory overflow

manaioussema, 2018-01-04:
Thank you for opening an issue on the Energia repository. 

### Do not use GitHub issues for project related questions. 

Use instead the [43oh forum](http://forum.43oh.com) and [TI E2E forum](http://e2e.ti.com), so everyone can share the questions and answers.

### Do not use GitHub issues for questions on third-party libraries. 

Use instead their own repositories.

### Check the issue hasn't been reported yet.

If an issue has alreday been reported, add your case to its thread.

### Please provide the following information.

* LaunchPad board name and version: `INSERT LAUNCHPAD BOARD NAME AND VERSION`
* Energia IDE version (found in Energia > About Energia menu): `INSERT ENERGIA VERSION`
* Board package version (found in Tools > Boards > Boards Manager menu): `INSERT BOARD PACKAGE VERSION`
* OS name and version: `INSERT OS NAME AND VERSION`

### Describe the expected results.

`INSERT DESCRIPTION OF THE EXPECTED RESULT`

### Describe the actual results.

`INSERT DESCRIPTION OF THE ACTUAL RESULT`

### List the steps to reproduce the problem.

`INSERT STEPS TO REPRODUCE THE PROBLEM`

robertinant, 2018-01-04:
You are using more flash space then is available. The only solution is to reduce the size of your Sketch. Note that the current compiler is only capable of utilizing the lower 48k. So even if the micro controller has more than 48k you will only be able to use 48k of that memory.

closed 2018-01-04 by robertinant

manaioussema, 2018-01-04:
any solution ?

rei-vilo, 2018-01-04:
Either use a LaunchPad with more flash or reduce the size of the executable.

For project-related questions, please post your question at the www.43oh.com forum.

manaioussema, 2018-01-04:
any idea about when the compiler will be updated ?

rei-vilo, 2018-01-04:
If your project is in a hurry, when don't you use CCS instead?

manaioussema, 2018-01-04:
i'm using a menu library that's not easy to port to CCS.