energia/Energia #998: CC3220SF analogRead() doe

RemusRTF, 2018-01-29:
Thank you for opening an issue on the Energia repository. 

### Do not use GitHub issues for project related questions. 

Use instead the [43oh forum](http://forum.43oh.com) and [TI E2E forum](http://e2e.ti.com), so everyone can share the questions and answers.

### Do not use GitHub issues for questions on third-party libraries. 

Use instead their own repositories.

### Check the issue hasn't been reported yet.

If an issue has alreday been reported, add your case to its thread.

### Please provide the following information.

* LaunchPad board name and version: CC3220SF
* Energia IDE version (found in Energia > About Energia menu): `1.6.1E18
* Board package version (found in Tools > Boards > Boards Manager menu): 5.6.0
* OS name and version: Windows10_1709

### Describe the expected results.

I wan`t to use analogRead to read out the analog voltage.

### Describe the actual results.

I get the error message: "undefined reference to `analogRead'"

### List the steps to reproduce the problem.

Use the analogRead() function on the CC3220SF and compile with Energia.
[image]

robertinant, 2018-01-29:
Please see resolution here:

https://e2e.ti.com/support/wireless_connectivity/simplelink_wifi_cc31xx_cc32xx/f/968/p/656105/2424837#2424837

"I have just pushed an update for the CC3220 in Energia. Please update to 5.6.1 using the board manager.

Note that A0 (pin 23) is shared with the UART pin that is connected to the emulator side (USB connection to the PC). If you would like to use A0 then you must RX jumper (J6) towards the CC3220SF. Note that you will not be able to send data using Serial to the PC when this jumper is switched. Rather then using A0, I would use A1 which is pin 25. This pin is exclusive to the BoosterPack header."

closed 2018-01-29 by robertinant

RemusRTF, 2018-01-29:
Thank you very much,
this removed the error but it caused another one to come up:
[image]