energia/Energia #1001: Nothing compiles on Energia 1.6.10E18 for MSP432

Akshay-Rs, 2018-02-04:
* LaunchPad board name and version: MSP432P401R
* Energia IDE version: 1.6.10E18 (Includes desktop binary and CCS Cloud IDE )
* Board package version (found in Tools > Boards > Boards Manager menu): The board package version is 3.8.0, but doesn't work for older versions either.
* OS name and version: Windows 10 Home 

Even basic example sketches (eg: Blinky) won't compile on Energia for the MSP432 Launchpad.
The error seems to be conflicting types for 'srandom'.
Here's a screenshot:
 
[image]

Akshay-Rs, 2018-02-04:
And yes, my Energia source is in a path with no spaces :)

robertinant, 2018-02-05:
There is an issue with arduino-builder, which Energia uses to build the Sketch, picking up the wrong compiler. If you have a RED MSP432 LaunchPad then install the "Energia MSP432 EMT RED boards" packet instead of the "Energia MSP432 boards" package. I will have a fix for the "Energia MSP432 boards" package later this week but if you have a RED board then use the RED board package.

closed 2018-02-05 by robertinant