energia/Energia #1005: CC3220 SLFS examples of WiFiMKII

vshymanskyy, 2018-02-15:
```
SLFS_String_Test.ino:16:18: fatal error: SLFS.h: No such file or directory
 #include <SLFS.h>
                  ^
compilation terminated.
Using library WiFiMKII at version 1.0.0 in folder: ...
```

Are there any plans to provide SLFS functionality to CC3220 series?
Are there any other libraries that enable SLFS access?

P.S. Also reported here: https://github.com/energia/WiFiMKII/issues/2

vshymanskyy, 2018-02-15:
I tried to use FS API directly, but unfortunately can't get it working.
I consulted with http://dev.ti.com/tirex/content/simplelink_cc32xx_sdk_1_40_00_03/docs/wifi_host_driver_api/html/group___file_system.html

```
#include <WiFi.h>
#include <ti/drivers/net/wifi/fs.h>

void setup() {
  Serial.begin(115200);
  delay(1000);
  
  char*           DeviceFileName = "MyFile.txt";
  unsigned long   MaxSize = 63 * 1024; //62.5K is max file size
  long            DeviceFileHandle = -1;
  _i32            RetVal = 0;        //negative retval is an error
  unsigned long   Offset = 0;
  unsigned char   InputBuffer[100] = {};
  _u32 MasterToken = 0;
  // Create a file and write data. The file in this example is secured, without signature and with a fail safe commit
  //create a secure file if not exists and open it for write.
  DeviceFileHandle =  sl_FsOpen((unsigned char *)DeviceFileName,
                       SL_FS_CREATE|SL_FS_OVERWRITE | SL_FS_CREATE_SECURE | SL_FS_CREATE_NOSIGNATURE | SL_FS_CREATE_MAX_SIZE( MaxSize ),
                       &MasterToken);
  Serial.println(String("sl_FsOpen ") + DeviceFileHandle);
  
  Offset = 0;
  //Preferred in secure file that the Offset and the length will be aligned to 16 bytes.
  RetVal = sl_FsWrite( DeviceFileHandle, Offset, (unsigned char *)"HelloWorld", strlen("HelloWorld")+1);
  Serial.println(String("sl_FsWrite ") + RetVal);
  
  RetVal = sl_FsClose(DeviceFileHandle, NULL, NULL , 0);

  Serial.println("File written");
  
  // open the same file for read, using the Token we got from the creation procedure above
  DeviceFileHandle =  sl_FsOpen((unsigned char *)DeviceFileName,
                                   SL_FS_READ,
                                   &MasterToken);
  Serial.println(String("sl_FsOpen ") + DeviceFileHandle);
  Offset = 0;
  RetVal = sl_FsRead( DeviceFileHandle, Offset, (unsigned char *)InputBuffer, strlen("HelloWorld")+1);
  Serial.println(String("sl_FsRead ") + RetVal);
  RetVal = sl_FsClose(DeviceFileHandle, NULL, NULL , 0);

  Serial.println("Content:");
  Serial.println((char*)InputBuffer);

}

void loop() {
  // put your main code here, to run repeatedly: 
  
}
```
This is basically copy-pasted from an example.
It builds and flashes OK, but then I get this output:
```
sl_FsOpen -2018
sl_FsWrite -2018
File written
sl_FsOpen -2018
sl_FsRead -2018
Content:
```
Have I missed something?

vshymanskyy, 2018-02-16:
Figured that out to some extent, but still no luck.
I added `WiFi.begin();` so the `sl_Start` gets called, now I get:
```
sl_FsOpen -10271
sl_FsWrite -10271
File written
sl_FsOpen -10271
sl_FsRead -10271
Content:
```
-10271 means `SL_ERROR_FS_PROGRAMMING_IN_PROCESS`.

The CC3220 Programmer's Guide says:
```
During the programming or restore to factory process, no file operation can be executed; when trying to read or write a file, an error of SL_ERROR_FS_PROGRAMMING_IN_PROCESS is received. In this case, the file system function can be re-invoked after the programming process is finished.
```
How do I know that the Programming process has finished?
Why is it happening in the first place? I'm using SOP configuration 000.

vshymanskyy, 2018-02-16:
@spirilis @robertinant @rei-vilo 
Could you take a look at this? I took SLFS part of CC3200 WiFi library and created a separate SLFS lib for CC3220. I need some help/answers to get it stable, and I think some approval to publish it.
Thanks.

Jack0wang, 2018-09-04:
@vshymanskyy , excuse me!
I have same question. Did you find the answers?
Thank you!

vshymanskyy, 2018-09-13:
@Jack0wang yes I did. it appears to be easier to use CCS instead of Energia

Jack0wang, 2018-09-14:
@vshymanskyy Unfortunately, I need it in Energia. 
Thank you for your replied.