energia/Energia #1009: CC1310 pin mapping

vonRaf, 2018-03-04:
Hi, where can I find pin mapping for CC1310? 
On Energia site there is missing link on image http://energia.nu/wordpress/wp-content/uploads/2016/11/LaunchPad-CC1310.png 
I try all combinations with DIO_XX, IO_XX  but without success. 

THX
Goran

rei-vilo, 2018-03-05:
See https://embeddedcomputing.weebly.com/launchpad-cc1310-sub-1-ghz.html

[image]

vonRaf, 2018-03-05:
Hi, THX for this, but on this I do not see name of, for example, DIO 15 pin that I can use in digitalWrite().

I try to include 
`#include <ti/devices/cc13x0/driverlib/ioc.h>
`
and then to use `IOID_15` and we will see will it work

Goran

rei-vilo, 2018-03-05:
Energia uses the pins numbers, _e.g._ 15,  not the pins names.

vonRaf, 2018-03-05:
hm, I will try that, I use PD_XX or PA_XX with Energia for Tiva C, so I thought that there is the same principe here
THX
Goran

rei-vilo, 2018-03-05:
From http://energia.nu/pin-maps/

> **Pin Naming**
> Users are encouraged to use the physical pin numbers (1-20, 1-40, or 1-80 depending on the LaunchPad) of the BoosterPack header pins as identifiers in their software.  Because these numbers are consistent between all Launchpads, this help to ensure the portability of the user’s software.  Some LaunchPads also support the use of GPIO or Port numbers such as “P1_3”.  These names are not recommended for use and will be deprecated in the future.  New LaunchPads that are added to Energia will not support pin names based on GPIO or Port numbers.

closed 2018-03-05 by rei-vilo

vonRaf, 2018-03-05:
:) you got me, I always forget RFM rule 

Sorry and THX
Goran

rei-vilo, 2018-03-05:
No problem, enjoy Energia!