energia/Energia #1011: Analogwrite not working on MSP432 on P6.5 and P6.4

fcooper, 2018-04-19:
Thank you for opening an issue on the Energia repository. 

### Do not use GitHub issues for project related questions. 

Use instead the [43oh forum](http://forum.43oh.com) and [TI E2E forum](http://e2e.ti.com), so everyone can share the questions and answers.

### Do not use GitHub issues for questions on third-party libraries. 

Use instead their own repositories.

### Check the issue hasn't been reported yet.

If an issue has alreday been reported, add your case to its thread.

### Please provide the following information.

* LaunchPad board name and version: MSP432 Red v2.1
* Energia IDE version (found in Energia > About Energia menu): 1.6.10E18
* Board package version (found in Tools > Boards > Boards Manager menu): 5.6.1
* OS name and version: Windows 10 64bit

### Describe the expected results.

PWM output on P6.4 and P6.5 when using analogWrite, The MSP432 pinmap indicates that those pins are valid for analogwrite.

### Describe the actual results.

No PWM output on pins P6.4 and P6.5. Verified by connecting logic analyzer to pins. When connected to those pins I see no output. When changing to other pins capable of analogwrite I see a square wave. Confirmed that the pins do work but toggling them using digitalWrites. 

### List the steps to reproduce the problem.

```
#define LED RED_LED
int left_pwm_pin = 10;  // P6.5

void setup() {                

pinMode(LED, OUTPUT);   
pinMode(left_pwm_pin, OUTPUT);      // sets the digital pin as output
analogWrite(left_pwm_pin, 125);
}

void loop() {
  digitalWrite(left_pwm_pin, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);               // wait for a second
  digitalWrite(left_pwm_pin, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);               // wait for a second
}
```

robertinant, 2018-11-14:
The pinmap is incorrect. P6.4/P6.5 are not PWM capable. Fixed with https://github.com/energia/energia.nu/commit/2b461aa5a99a750f8d3b102adbf2551ee2dd02f5

closed 2018-11-14 by robertinant

rei-vilo, 2018-11-14:
Updated pins map

[image]