energia/Energia #1014: Energia MT: Monitor example gives obscure output

armsp, 2018-06-02:
I am using the MSP432 EMT RED Board on Windows 10 with the latest version of Energia.
I just ran the **Monitor** example in Multitasking and it gave me the following output on the serial monitor -
```
ÿÿÿÿÿÿÿ
Command too long.
> ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
Command too long.
> ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
Command too long.
> ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
Command too long.
> ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
Command too long.
> ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
Command too long.
> ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
```
and so on and on......
I believe this isn't supposed to happen. Any information on how to resolve it and to show the correct stats.

NOTE: _As mentioned in the sketch i set the baud rate at 9600 so i don't think incorrect baud rate is the issue._

Also, since this is a fairly complex sketch, i wanted to know where do i look to understand it fully. I can't seem to find a proper documentation for all the functions used and libraries used in the sketch. Please guide me in the right direction.

Thankyou

rei-vilo, 2018-06-02:
+ Try against the non-red `LaunchPad w/ MSP432 EMT (48 MHZ)` with boards package `Energia MSP432 boards 3.8.1`.

[image]

+ Find useful information and pointers at [LaunchPad MSP432: RTOS for Everyone](https://embeddedcomputing.weebly.com/launchpad-msp432-rtos-for-everyone.html) and [Exploring RTOS with Galaxia on Energia MT](https://embeddedcomputing.weebly.com/exploring-rtos-with-galaxia.html).

armsp, 2018-06-02:
Since my Launchpad is actually a RED colored one, shouldn't i be using the option that i have already been using?
`Try against the non-red LaunchPad w/ MSP432 EMT (48 MHZ) with boards package Energia MSP432 boards 3.8.1.`
Sadly that doesn't work either. I get an `exit status 1`. Following is what gets displayed -

Energia: 1.6.10E18 (Windows 10), Board: "LaunchPad w/ msp432 EMT (48MHz)"

Build options changed, rebuilding all
C:\Users\Dell\AppData\Local\Energia15\packages\energia\hardware\msp432\3.8.0\cores\msp432\ti\runtime\wiring\random.c:83:6: error: conflicting types for 'srandom'

 void srandom(unsigned long seed)

      ^~~~~~~

In file included from c:\users\dell\appdata\local\energia15\packages\energia\tools\arm-none-eabi-gcc\6.3.1-20170620\arm-none-eabi\include\stdlib.h:11:0,

                 from C:\Users\Dell\AppData\Local\Energia15\packages\energia\hardware\msp432\3.8.0\cores\msp432\ti\runtime\wiring\random.c:42:

c:\users\dell\appdata\local\energia15\packages\energia\tools\arm-none-eabi-gcc\6.3.1-20170620\arm-none-eabi\include\stdlib.h:257:7: note: previous declaration of 'srandom' was here

 _VOID _EXFUN(srandom,(unsigned));

       ^

exit status 1
Error compiling for board LaunchPad w/ msp432 EMT (48MHz)

robertinant, 2018-06-04:
There are 2 options to fix this. 

1. In mon_setup() call `Serial.begin(115200, true);` instead of `Serial.begin(115200);` This will put the Serial port in blocking read mode.
2. On line 144 of the monitor Sketch just before `Serial.read()` and after `while (true)` put the following:
```
if(!Serial.available()) {
    Task_yield();
}
```        
It should then read:
```
int escape_index = 0;

while (true) {
    if(!Serial.available()) {
        Task_yield();
        continue;
    }

char c = Serial.read();
```

closed 2018-06-04 by robertinant

rei-vilo, 2018-06-04:
Is the example going to be updated accordingly so the next release will include the fix?

armsp, 2018-06-05:
@robertinant @rei-vilo i did as you mentioned, but then it just acts like an echo prorgram. According to the documentation here http://energia.nu/tag/energia-mt/ it should show some stats about CPU load-
`Monitor: This is an awesome utility Sketch that displays CPU utilization, task memory usage, etc. There are a couple of VT100 features in this sketch so best would be to use a VT100 terminal rather than the build in Serial monitor. For example, to continuously display the CPU load, type stats <space> <enter> and the CPU load will be updated in the terminal continuously.`

But that doesn't happen.
Currently i am getting the following output (after typing **_stats_**) - 
```
Welcome! This is the Serial debug console.
> stats 
```