energia/Energia #1015: driverlib library is out-of-date for  MSP-EXP430FR2433

bootchk, 2018-06-13:
Issue has been reported in other old threads?

Context:
 MSP-EXP430FR2433
energia-1.6.10E18
MSP-EXP430FR2433LP
Ubuntu17.04

Expected: to be able to use latest MSPWare DriverLib. 

Actual results: for example, one of many compile errors:   error: 'PMM_RST_INTERRUPT' was not declared in this scope

Steps to reproduce the problem.
1. cut and paste into Energia:  msp430ware_3_80_04_05/driverlib/examples/MSP430FR2xx_4xx/pmm/pmm_Reset_LPMx_5.c
2. copy MSPWare...Board.h into a new Energia library say at ~/Energia/libraries/Board
3. Choose "Verify" icon.

That example has:
#include "driverlib.h"
#include "Board.h"

One problem is that the driverlib.h in the Energia distribution has many ifdefs, but none for __MSP430FR2433__.  I worked around that by defining __MSP430FR5969__ (which is referenced in driverlib.h)  That seems to resolve many undefined things things (although I am not sure the definitions are correct for MSP430fr2433) but did not resolve definitions for the PMM module, which driverlib does not seem to include under any circumstances.  Also, as a workaround, defining __MSP430FR5969__ might be fraught with complexity, since other macro definitions (such as from MSPWare...Board.h) might depend on__MSP430FR2433__ being defined?

I suppose the Energia driverlib support is not intended to fully support all of MSPWare driverlib?

Maybe I should include MSPWare driverlib.h instead of Energia driverlib.h? I.E. its a filename clash that I can resolve by using a full path?

bootchk, 2018-06-13:
I tried copying MSPWare.../driverlib/driverlib/MSP430FR2xx_4xx (a directory) to ~/Energia/libraries  (making a new Energia library.)   Then I copied assert.h from Energia (which ignores assert statements) into the library.  Then Energia would compile said MSPWare example. 

Loading and running,  I think it produced the expected results: LED's were both on at startup.  I think that in Energia, after loading  Energia makes the debugger probe assert notRST/NMI pin, then the program then lights an LED and proceeds to light another LED and sleep LPM3.5.

I am just reporting this for other readers, who might want to use MSPWare examples on Energia.

I am really looking for an example for the MSP-EXP430FR2433 that enters and exits LPM4.5 on a button press.  I will take that up in another forum.

rei-vilo, 2018-06-13:
> I am really looking for an example for the MSP-EXP430FR2433 that enters and exits LPM4.5 on a button press. I will take that up in another forum.

Please give a look at [Ultra-Low Power with MSP430](https://embeddedcomputing.weebly.com/ultra-low-power-with-msp430.html).