energia/Energia #1018: Bad mspdebug command "No such file or directory"

AlexusBlack, 2018-10-03:
Upload action calls wrong mspdebug command

* LaunchPad board name and version: MSP430G2 with MSP430G2553
* Energia IDE version (found in Energia > About Energia menu): 1.8.7E20
* Board package version (found in Tools > Boards > Boards Manager menu): 1.0.4
* OS name and version: Mac OS X 10.13.6

### Describe the expected results.

Upload action calls
`/Applications/Energia.app/Contents/Java/hardware/tools/mspdebug/mspdebug rf2500 --force-reset 'prog /var/folders/b_/7vlctxfj04nd4g_z2r7y3vx40000gn/T/arduino_build_59286/sketch_oct03a.ino.hex'`
(prog and file path enclosed with quotes together)
that uploads to MC without errors

### Describe the actual results.
Upload action calls mspdebug with "prog" outside of quotes what results in error in Energia and when command copied in terminal
```
Energia: 1.8.7E20 (Mac OS X), Board: "MSP-EXP430G2 w/ MSP430G2553"

Build options changed, rebuilding all
Sketch uses 816 bytes (4%) of program storage space. Maximum is 16384 bytes.
Global variables use 20 bytes (3%) of dynamic memory, leaving 492 bytes for local variables. Maximum is 512 bytes.
/Applications/Energia.app/Contents/Java/hardware/tools/mspdebug/mspdebug rf2500 --force-reset prog '/var/folders/b_/7vlctxfj04nd4g_z2r7y3vx40000gn/T/arduino_build_59286/sketch_oct03a.ino.hex' 
MSPDebug version 0.22 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2013 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Trying to open interface 1 on 013
Initializing FET...
FET protocol version is 30394216
Set Vcc: 3000 mV
Configured for Spy-Bi-Wire
Sending reset...
Device ID: 0x2553
  Code start address: 0xc000
  Code size         : 16384 byte = 16 kb
  RAM  start address: 0x200
  RAM  end   address: 0x3ff
  RAM  size         : 512 byte = 0 kb
Device: MSP430G2xx3
Number of breakpoints: 2
fet: FET returned NAK
warning: device does not support power profiling
Chip ID data: 25 53
prog: '/var/folders/b_/7vlctxfj04nd4g_z2r7y3vx40000gn/T/arduino_build_59286/sketch_oct03a.ino.hex': No such file or directory
An error occurred while uploading the sketch
```

### List the steps to reproduce the problem.

Press upload button.

Problem can be easily fixed by moving singular quote before file name to be before "prog" command.

AlexusBlack, 2018-10-10:
The problem seams to be in the Java/hardware/energia/msp430/platform.txt file on lines 112 and 117. There are single quotes around file name and they break the command (no sure if on MacOSX only).
Removing single quotes from around the .hex file makes in-energia program\upload possible again.

ponyatov, 2019-09-19:
I use mspdebug directly (Debian Linux) it requires single right-down quoted to run 'prog file.hex' but the programming speed is very very low (EXP-MSP430G2 Launchpad + MSP430F5418A).

ajitjadhav28, 2020-03-09:
> The problem seems to be in the Java/hardware/energia/msp430/platform.txt file on lines 112 and 117. There are single quotes around file name and they break the command (no sure if on MacOSX only).
> Removing single quotes from around the .hex file makes in-energia program\upload possible again.

Worked! :)

energia-1.8.10E23
Ubuntu 18.04

RamonTav, 2020-09-03:
> The problem seems to be in the Java/hardware/energia/msp430/platform.txt file on lines 112 and 117. There are single quotes around file name and they break the command (no sure if on MacOSX only).
> Removing single quotes from around the .hex file makes in-energia program\upload possible again.

It  Works!!!
/energia-1.8.10E23-linux64/energia-1.8.10E23/hardware/energia/msp430/platform.txt
MSP-EXP430G2
Fedora 32
Muchas gracias!

mhunsberger56, 2020-10-24:
> The problem seems to be in the Java/hardware/energia/msp430/platform.txt file on lines 112 and 117. There are single quotes around file name and they break the command (no sure if on MacOSX only).
> Removing single quotes from around the .hex file makes in-energia program\upload possible again.

DUDE!!

You rock.  I knew it had to be something simple.  Thank you for saving me some digging.

foxxmotor, 2021-02-14:
> The problem seems to be in the Java/hardware/energia/msp430/platform.txt file on lines 112 and 117. There are single quotes around file name and they break the command (no sure if on MacOSX only).
> Removing single quotes from around the .hex file makes in-energia program\upload possible again.

It's also work for me. Thank you so much!

fatimaaijaz1212, 2024-02-06:
> It's also work for me. Thank you so much!

heyy, how exactly did you implement it that it works? Need help ASAP please do let me know