energia/Energia #1025: MSP432E and TM4C129 EMT — Hard-Coded Variables

rei-vilo, 2018-10-21:
Why `-DBOARD_MSP_EXP432E401Y` and `-D__MSP432E401Y__` are hard-coded in `platform.txt`, 

```
build.extra_flags=-mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mabi=aapcs -g -Dxdc_target_types__=gnu/targets/arm/std.h -Dxdc_target_name__=M4F -Dxdc_cfg__xheader__="configPkg/package/cfg/energia_pm4fg.h" -DBOARD_MSP_EXP432E401Y -Dxdc__nolocalstring=1 -D__MSP432E401Y__
```

even for the **EK-TM4C1294XL** board, for which `boards.txt` provides

```
EK-TM4C1294XL.build.variant=EK_TM4C1294XL
EK-TM4C1294XL.build.board=EK_TM4C1294XL
```

robertinant, 2018-10-22:
D__MSP432E401Y__ has to be defined since the SDK requires it even though we compile for the TIVAC board. Essentially, there is no difference between the 2 boards. I have changed -DBOARD to specify the TIVAC when compiling for TIVAC.

robertinant, 2018-10-22:
I take that back. The TivaC launchpad in this package has to be treated as if it was a MSP432E launchpad and hence the defines have to be as they are.

closed 2018-10-22 by robertinant