energia/Energia #1028: MSP430FR2433 SleepSeconds high power usage

jgrrule, 2018-10-23:
Thank you for opening an issue on the Energia repository. 

### Do not use GitHub issues for project related questions. 

Use instead the [43oh forum](http://forum.43oh.com) and [TI E2E forum](http://e2e.ti.com), so everyone can share the questions and answers.

### Do not use GitHub issues for questions on third-party libraries. 

Use instead their own repositories.

### Check the issue hasn't been reported yet.

If an issue has alreday been reported, add your case to its thread.

### Please provide the following information.

* LaunchPad board name and version: `MSP-EXP430FR2433`
* Energia IDE version (found in Energia > About Energia menu): `18 & 20'
* Board package version (found in Tools > Boards > Boards Manager menu): `1.0.5`
* OS name and version: `Windows 10

### Describe the expected results.

SleepSeconds puts the device into a standby state with a current draw of approx 1ua

### Describe the actual results.

Current when device is in sleep seconds is 18ua. Verified with a ucurrent Gold and also tested using both CodeComposerStudio and Energia IDE. 

When not using Energia, and using TI CCS examples; device operates at <1ua in LPM3

### List the steps to reproduce the problem.

Create a sketch, initialize all GPIO as INPUT_PULLDOWN. Put device into SleepSeconds(anynumber).

jgrrule, 2018-11-07:
Hi, this issue is still present and I have had no luck getting the consumption any lower.

Any advice would be greatly appreciated.

RobPo, 2018-11-08:
Hello, have seen the same issue here.  A low sleep current was my main motivation to try MSP430 platform. Any support would be great.

StefanSch, 2018-11-08:
Hi, currently i do not have access to a FR2433 Launchpad but when looking into the code i think one line is missing during the XTAL startup.
You can try easily yourself the fix when adding this line into the setup function:
		CSCTL4 |= SELA__XT1CLK;
Would be great to get feedback if this helps.

RobPo, 2018-11-08:
Hi, unfortunately not. Using this minimal example

```
void setup() {
     for (int i = 2; i < 20; i++) {
       pinMode(i, OUTPUT);
       digitalWrite(i, LOW);
     }
    CSCTL4 |= SELA__XT1CLK;
    sleepSeconds(500);
}

void loop() {
}
```
Energytrace still measures 18µA once the programming jumpers are removed.

StefanSch, 2018-11-08:
Sorry, it should be:
		CSCTL4 &= ~SELA__REFOCLK;
REFO Selection bit needs to be cleared.

RobPo, 2018-11-08:
Nope, still consuming 18µA.

StefanSch, 2018-11-08:
OK, thanks for testing - will try to grab a FR2433 Launchpad in debug into that in the next few days.

RobPo, 2018-11-08:
Thanks that would be great...

StefanSch, 2018-11-12:
When doing some testing and looking closer into the Launchpad layout i could see that the crystal is mounted by not connected by default with the 0 Ohm resistors R2 and R3.
So what in this case happens is that the REFOSC is used internally for the 32kHz clock which does consume the ~17uA more then when using the 32kHz Crystal oscillator.
So i am just now wondering who you could get the lower current when using the example.
May you can let me know if you modified the board and which example you used for that.

So finally it is easy to get the current down but requires to change some 0 Ohm Resistors and a small fix on the Energia lib - will do some tests and updates once everything is clear.

Thanks.

RobPo, 2018-11-12:
OK thanks. I just moved the 0 Ohm resistors from R4/R5 (default; no external crystal) to R2/R3 (external crystal selected) on the launchpad. 

```
void setup() {
     for (int i = 2; i < 20; i++) {
       pinMode(i, OUTPUT);
       digitalWrite(i, LOW);
     }
    CSCTL4 &= ~SELA__REFOCLK;
    sleepSeconds(500);
}

void loop() {
}
```
Using this example the current consumption is unchanged at ~18µA.

StefanSch, 2018-11-12:
This code changes the configuration of the pins where the crystal is connected.
So pin 11 and pin 12 needs to be excluded.
With changing the Resistor setting in with below code i can measure 0,0009 mA with the Energy Trace.
(The startup code for the Crystal is OK but could be structured better.)

void setup() {
     for (int i = 2; i < 20; i++) {
       if (!(i == 11 || i == 12)){
           pinMode(i, OUTPUT);
           digitalWrite(i, LOW);
       }
     }
    sleepSeconds(500);
}

void loop() {
}

jgrrule, 2018-11-13:
Hi, can I just confirm this means that <18ua for SleepSeconds is only possible when using the external crystal? And an internal clock can not be used to achieve this?

StefanSch, 2018-11-13:
Currents below 1uA can be reached with this device but with the structure currently used in Energia this is quite complicated and would require several exceptions in the software flow.
When doing a native code below example shows how this can be done using the VLO to trigger the WDT Timer regularly wake up the CPU
`#include <msp430.h>

int main(void)
{

    WDTCTL = (WDT_ADLY_250 & ~(WDTSSEL__ACLK)) | WDTSSEL__VLO;
    SFRIE1 |= WDTIE;                        // Enable WDT interrupt

    // Port Configuration all un-used pins to output low
    P1OUT = 0x00;
    P2OUT = 0x00;
    P3OUT = 0x00;
    P1DIR = 0xff;
    P2DIR = 0xff;
    P3DIR = 0xff;

    // Disable the GPIO power-on default high-impedance mode
    // to activate previously configured port settings
    PM5CTL0 &= ~LOCKLPM5;

    CSCTL3 = SELREF__REFOCLK;
    CSCTL4 = SELA__REFOCLK | SELMS__VLOCLK;
    // XT1 is off so flag should be cleared
    CSCTL7 &= ~(XT1OFFG | DCOFFG);      // Clear XT1 and DCO fault flag
    SFRIFG1 &= ~OFIFG;
    SFRIE1 |= WDTIE;                        // Enable WDT interrupt

    __bis_SR_register(LPM4_bits | GIE);     // Enter LPM4 - WDT still running from VLO
    __no_operation();                       // For debug

}

// Watchdog Timer interrupt service routine
#if defined(__TI_COMPILER_VERSION__) || defined(__IAR_SYSTEMS_ICC__)
#pragma vector=WDT_VECTOR
__interrupt void WDT_ISR(void)
#elif defined(__GNUC__)
void __attribute__ ((interrupt(WDT_VECTOR))) WDT_ISR (void)
#else
#error Compiler not supported!
#endif
{
    P1OUT ^= BIT0;                          // Toggle P1.0 (LED) every 1s
}
`

closed 2018-11-13 by StefanSch