energia/Energia #1036: Can't compile anything: Windows 32 bit Issue

fcooper, 2018-11-26:
* LaunchPad board name and version: MSP-EXP432P401R
* Energia IDE version (found in Energia > About Energia menu): 1.8.7E21
* Board package version (found in Tools > Boards > Boards Manager menu): MSP432 EMT RED boards v5.23.1 (I believe)
* OS name and version: Windows (Version TBD) Likely 32bit version

Compiling Blink program is failing. Likely is an issue with 32bit windows.

Verbose Output Error Message:
``
C:\ti\energia-1.8.7E21_2\energia-1.8.7E21\arduino-builder -dump-prefs -logger=machine -hardware C:\ti\energia-1.8.7E21_2\energia-1.8.7E21\hardware -hardware C:\Users\Bob Newhart\AppData\Local\Energia15\packages -tools C:\ti\energia-1.8.7E21_2\energia-1.8.7E21\tools-builder -tools C:\ti\energia-1.8.7E21_2\energia-1.8.7E21\hardware\tools\msp430 -tools C:\Users\Bob Newhart\AppData\Local\Energia15\packages -built-in-libraries C:\ti\energia-1.8.7E21_2\energia-1.8.7E21\libraries -libraries C:\Users\Bob Newhart\Documents\Energia\libraries -fqbn=energia:msp432r:MSP-EXP432P401R -ide-version=10807 -build-path C:\Users\BOBNEW~1\AppData\Local\Temp\arduino_build_894357 -warnings=none -build-cache C:\Users\BOBNEW~1\AppData\Local\Temp\arduino_cache_460968 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.dslite.path=C:\Users\Bob Newhart\AppData\Local\Energia15\packages\energia\tools\dslite\8.2.0.1400 -prefs=runtime.tools.dslite-8.2.0.1400.path=C:\Users\Bob Newhart\AppData\Local\Energia15\packages\energia\tools\dslite\8.2.0.1400 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Bob Newhart\AppData\Local\Energia15\packages\energia\tools\arm-none-eabi-gcc\6.3.1-20170620 -prefs=runtime.tools.arm-none-eabi-gcc-6.3.1-20170620.path=C:\Users\Bob Newhart\AppData\Local\Energia15\packages\energia\tools\arm-none-eabi-gcc\6.3.1-20170620 -prefs=runtime.tools.ino2cpp.path=C:\Users\Bob Newhart\AppData\Local\Energia15\packages\energia\tools\ino2cpp\1.0.4 -prefs=runtime.tools.ino2cpp-1.0.4.path=C:\Users\Bob Newhart\AppData\Local\Energia15\packages\energia\tools\ino2cpp\1.0.4 -verbose C:\ti\energia-1.8.7E21\examples\01.Basics\Blink\Blink.ino

processing.app.debug.RunnerException
	at cc.arduino.Compiler.callArduinoBuilder(Compiler.java:325)
	at cc.arduino.Compiler.loadPreferences(Compiler.java:216)
	at cc.arduino.Compiler.build(Compiler.java:175)
	at processing.app.SketchController.build(SketchController.java:664)
	at processing.app.Editor$BuildHandler.run(Editor.java:1616)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: Cannot run program "C:\ti\energia-1.8.7E21_2\energia-1.8.7E21\arduino-builder": CreateProcess error=216, This version of %1 is not compatible with the version of Windows you're running. Check your computer's system information and then contact the software publisher
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.Compiler.callArduinoBuilder(Compiler.java:305)
	... 5 more
Caused by: java.io.IOException: CreateProcess error=216, This version of %1 is not compatible with the version of Windows you're running. Check your computer's system information and then contact the software publisher
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 7 more
``

fcooper, 2018-11-26:
Also I know there is a space in the AppData and Documents path because of the user "Bob Newhart". But I wasn't sure if this is the same space in name error or a in2cpp error since the error message looks different.

mallikarjunasai995, 2018-12-15:
though the spaces are not there in the username still throwing an error for E21... previous version E18 is working fine

Bundaboy53, 2020-02-10:
Did anyone get to the bottom of this issue?