energia/Energia #1042: MSP430 micros() accuracy is unusable

Schuwi, 2019-07-15:
* LaunchPad board name and version: MSP-EXP430G2 v1.5
* Energia IDE version (found in Energia > About Energia menu): v1.8.7E21
* Board package version (found in Tools > Boards > Boards Manager menu): v1.0.5
* OS name and version: Windows 10

Since commit 8eb738e86ae2763ffa242a08f425391986dcd09a the resolution of the micros() function is 512μs on a 16MHz clock (tested). If I understand the code correctly the resolution is even worse at 8MHz clock with 1024μs, so actually coarser than the millis() function.
I use the MSP430G2553 to receive RF signals, which is basically impossible to do with this resolution.
I reverted the code affecting micros() to the state before the commit in my local Energia installation as a workaround.

TwoRedCells, 2020-07-01:
I am confirming that I get 1024μs resolution on MSP430FR2433 @ 8MHz.
Workaround is to build your own with a timer. :|

REGIOIGER, 2022-04-20:
Hi Shuwi, Im interested in know how to revert the code affecting micros(), is necessary re-compiling ENERGIA IDE?

Schuwi, 2022-04-20:
As this was quite some time ago and apparently I updated the board library in the meantime (or reinstalled my whole system) my local changes were gone.
But I did some digging and think I know what I did:
1. Find where Energia saves the board libraries (those in "Tools -> Board -> Board-Manager..."). For me on linux it's `/home/schuwi/.energia15/packages/energia/hardware`.
2. Edit the wiring.c in `msp430` (`/home/schuwi/.energia15/packages/energia/hardware/msp430/1.0.7/cores/msp430/wiring.c` for me).
3. Revert [this change](https://github.com/energia/Energia/commit/8eb738e86ae2763ffa242a08f425391986dcd09a#diff-fd93722d5a19678473b467a3877c64d771ed8c2afcfd5a81578ffcfa5dbefae3L34-L36). So that in the end `TICKS_PER_WDT_OVERFLOW` has the value 512 again, as before the commit in question.

No need to recompile Energia :)

REGIOIGER, 2022-04-20:
Thanks a lot Schuwi, I was editing in the wrong place, you helped me a lot  :)