energia/Energia #1043: Launch Pad Tiva C Tm4c129   SPI clock

ChubbyBunny1, 2019-09-23:
LaunchPad board name and version: Launch Pad Tiva C Tm4c129 120mhz
 Energia IDE version (found in Energia > About Energia menu): 1.8.7E21
 Board package version (found in Tools > Boards > Boards Manager menu): 1.0.3

OS Windows10

Hi, the default speed for the SPI is 4mhz, is any way to get higher speeds with Energia? I am controlling a lcd tft 3.5 display with ili9486 driver and i want to get higher refresh rates.
I suppose we can modify the energia spi library or something similar to achive this.

rei-vilo, 2019-09-23:
See #470, #489 &nd #1003.

ChubbyBunny1, 2019-09-24:
Thanks for your responce, seems i cant find the path where the spi.h is located. I searched all the subfolders of energia ide. I can find only the msp430 spi.h file.
Sorry for posting this here, it would be great if you can help

StefanSch, 2019-09-26:
I guess you are struggling with searching at the right place.
As the TivaC is an package which is loaded after the installation, it is not in the default energia folder but in the package folder. For windows this is 
C:\Users\USERNAME\AppData\Local\Energia15\packages\energia\hardware\tivac
and you then find spi.h here:
c:\Users\USERNAME\AppData\Local\Energia15\packages\energia\hardware\tivac\1.0.3\libraries\SPI\SPI.h
For MAC and Linux this is similar but i do not know the path by heart.

ChubbyBunny1, 2019-09-26:
I already found the directory by pressing CLTR+K, it takes you to C:\Users\USERNAME\AppData\Local\Energia15\packages\energia\hardware\tivac
then you can find everything there.

In the SPI.cpp file, to modify the SPI clock the line 201 needs to be changed accordingly.

ROM_SSIConfigSetExpClk(SSIBASE, F_CPU, SSI_FRF_MOTO_MODE_0, SSI_MODE_MASTER, 4000000, 8);
eg. chaning the default 4000000 (4mhz) number to 8000000 (8mhz).

According to tivaware peripherial driver library pdf we can see the command parameters
SSIConfigSetExpClk(uint32_t ui32Base,
uint32_t ui32SSIClk,
uint32_t ui32Protocol,
uint32_t ui32Mode,
uint32_t ui32BitRate,
uint32_t ui32DataWidth)

Essam-droid, 2020-03-02:
> LaunchPad board name and version: Launch Pad Tiva C Tm4c129 120mhz
> Energia IDE version (found in Energia > About Energia menu): 1.8.7E21
> Board package version (found in Tools > Boards > Boards [...]

Hello ChubbyBunny1,

I've noticed that you are using ILI9486 driver with Tiva C launchPad.
I experience an issue that RGB colors are not displayed on the touch screen, only black,white and gray are displayed, Have you stucked in this issue before?

thank you.

ChubbyBunny1, 2020-03-18:
> Hello ChubbyBunny1,
>
> I've noticed that you are using ILI9486 driver with Tiva C launchPad.
> I experience an issue that RGB colors are not displayed on the touch screen, only black,white and gray are [...]

Hello, i didnt play much around with ISP with this driver, i moved to 16bit parallel interface which required to solder 16 wires on the lcd and writing the library again. Maybe you can check your wiring, but should be a coding problem. This problem occured on me on the parallel interface, should be a problem on bit order, how you send the information to the lcd. I can share the library that i modified for 16 bit parallel interface.