energia/Energia #1044: Sketches run very slowly on M430G2452 when compiled/uploaded from Energia 1.8.7E21

reppad, 2019-10-03:
* LaunchPad board name and version: `MSP-EXP430G2 rev 1.5` (with M430G2452 chip)
* Energia IDE version (found in Energia > About Energia menu): `Energia 1.8.7E21`
* Board package version (found in Tools > Boards > Boards Manager menu): `Energia MSP430 boards version 1.0.5`
* OS name and version: `Ubuntu 18.04.3 LTS`

### General description

When I upload a sketch on my launchpad board with MSP430G2452 chip (not tested with 2553), the program runs very slowly.
This behaviour is very visible on the blink example, the LED turns ON or OFF after long time (something like 10 seconds).
When I upload the same sketch on the the same board with energia 0101E0016, there is no problem, the LED state change each second.

### Describe the expected results.

Program running at normal speed when compiled/uploaded from Energia 1.8.7E21

### Describe the actual results.

The program runs very slowly when compiled/uploaded from Energia 1.8.7E21

### List the steps to reproduce the problem.

1. connect the board to an USB port
2. open the blink example (or every sketch)
3. select the correct board and port in tools menu if needed
4. clic on upload button

StefanSch, 2019-10-04:
Hi, 
this is an issue which was fixed already a while ago but as there was no release in the mean time not in the release package so far - but i will go into the next version.
If you would like to fix it know just take the wiring.c file out of the repository and replace it with your local version.
https://github.com/energia/msp430-lg-core/blob/master/cores/msp430/wiring.c

reppad, 2019-10-04:
Thank you very much, it works perfectly now !

closed 2019-10-04 by reppad