energia/Energia #1047: PROBLEM WITH SERIAL CONNECTION IN MAC OS CATALINA

cenkaybiyik, 2019-10-24:
Dear﻿ All, 

I have a problem with my Energia IDE. I have using my Tiva c ek-tm4c123gxl board. Also I am using the Energia ide on Mac OS Catalina 10.15. But in Energia IDE in the Mac doesn't see my board. I cannot see anything except for bluetoothport in serial port place.On the other hand I don't have any problems with Arduino boards﻿ and the Arduino ide in Mac too. Help me with this problem please.Have a nice day and good luck :)﻿

Thank you for opening an issue on the Energia repository. 

### Do not use GitHub issues for project related questions. 

Use instead the [43oh forum](http://forum.43oh.com) and [TI E2E forum](http://e2e.ti.com), so everyone can share the questions and answers.

### Do not use GitHub issues for questions on third-party libraries. 

Use instead their own repositories.

### Check the issue hasn't been reported yet.

If an issue has alreday been reported, add your case to its thread.

### Please provide the following information.

* LaunchPad board name and version: `INSERT LAUNCHPAD BOARD NAME AND VERSION`
* Energia IDE version (found in Energia > About Energia menu): `INSERT ENERGIA VERSION`
* Board package version (found in Tools > Boards > Boards Manager menu): `INSERT BOARD PACKAGE VERSION`
* OS name and version: `INSERT OS NAME AND VERSION`

### Describe the expected results.

`INSERT DESCRIPTION OF THE EXPECTED RESULT`

### Describe the actual results.

`INSERT DESCRIPTION OF THE ACTUAL RESULT`

### List the steps to reproduce the problem.

`INSERT STEPS TO REPRODUCE THE PROBLEM`

robertinant, 2019-11-11:
Please continue the discussion here in the tiva-core repo: https://github.com/energia/tivac-core/issues/44

robertinant, 2019-11-11:
I don't see any issues either so this is not a bug. Can you please post a question in the forum here: https://forum.43oh.com/forum/125-energia-tivaccc3xxx/ so that we can follow up there:

closed 2019-11-11 by robertinant