energia/Energia #1050: MSP430F5529 can't upload

MrThanlon, 2019-12-07:
* LaunchPad board name and version: MSP430F5529
* Energia IDE version (found in Energia > About Energia menu): 1.8.10E22
* Board package version (found in Tools > Boards > Boards Manager menu): 1.0.5
* OS name and version: macOS 10.15

### Describe the expected results.

```
Done uploading.
```

### Describe the actual results.

```
___REMOVE___/DebugServer/bin/DSLite load -c ___REMOVE___/MSP-EXP430F5529LP.ccxml -f /var/folders/ty/hy53t_293mg6_t51snqngp_00000gn/T/arduino_build_827602/sketch_dec07a.ino.elf 
java.io.IOException: Cannot run program "___REMOVE___/DebugServer/bin/DSLite": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.runCommand(SerialUploader.java:383)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:197)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2075)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
An error occurred while uploading the sketch
```

### List the steps to reproduce the problem.
[image]

It seems IDE got the wrong path of DSLite? But it works on windows version.

shuchow, 2019-12-10:
I'm having the same exact problem also on Catalina.  Board is a MSP430G2ET.

robertinant, 2019-12-10:
This is indeed a bug. The wrong version of DSLite was included in Energia 22. I will have a new release (Energia 23) later this week. For now, use Energia 21.
Energia 23 will also include a new release of the MSP430 core.

rei-vilo, 2019-12-10:
See **Suggestion: Consolidate tool-chains and utilities #1051** at https://github.com/energia/Energia/issues/1051

shuchow, 2019-12-11:
> This is indeed a bug. The wrong version of DSLite was included in Energia 22. I will have a new release (Energia 23) later this week. For now, use Energia 21.
> Energia 23 will also include a new release of the MSP430 core.

Great, thank you!

Is there an official place to download compiled 23?  Energia.nu most recent link is to 1.8.7E21.  I stumbled upon 1.8.10E22 from your post on 43oh.com's forums, but was wondering if there was a more official place to find links to early releases.  Thanks!

MrThanlon, 2019-12-11:
> Great, thank you!
>
> Is there an official place to download compiled 23? Energia.nu most recent link is to 1.8.7E21. I stumbled upon 1.8.10E22 from your post on 43oh.com's forums, but was wondering if [...]

I modified the `platform.txt` file(`/Applications/Energia.app/Contents/Java/hardware/energia/msp430/platform.txt`), it works currently.
On line 96:
```
...
tools.dslite.path={runtime.tools.dslite-8.2.0.1400.path}
tools.dslite.config.path={path}
tools.dslite.cmd.path={runtime.ide.path}/hardware/tools/DSLite/DebugServer/bin/DSLite
tools.dslite.upload.pattern={cmd.path} {upload.verbose} load -c "{runtime.ide.path}/hardware/energia/msp430/variants/{build.variant}/{build.variant}.ccxml" -f "{build.path}/{build.project_name}.elf"
...
```
It's not the best solution, but you can use it for emergency. Or you can try to compile Energia with latest msp430-core.

shuchow, 2019-12-11:
@MrThanlon Thanks, your suggestion got me closer, but it looks like I have a problem specific to my board.  When I try to upload my code, I get:

`
Failed: Cannot read System Setup data from XML file /Applications/Energia.app/Contents/Java/hardware/energia/msp430/variants/MSP-EXP430G2ET2553LP/MSP-EXP430G2ET2553LP.ccxml
`

Looking it that EXP430G2ET2553LP directory, the ccxml file is named MSP-EXP430G2553LP.ccxml, which is a different basename than the parent directory name, causing the read error.  If I rename the file to EXP430G2ET2553LP.ccxml, I get a dreaded "No USB FET was found" error.

trick2011, 2020-05-11:
@robertinant The noted problem seems to persist into Energia 23. Energia 21 runs perfectly from download but 23 doesn't and gives a familiar: 
```
java.io.IOException: Cannot run program "{runtime.tools.dslite-9.2.0.1793-e1.path}/DebugServer/bin/DSLite": error=2, No such file or directory
```
I'm using the Linux version.

jackakyle, 2020-07-17:
@trick2011 and @robertinant just wanted to confirm that I also have the same issue, with Energia 21 working but 23 giving me the same error of 

`java.io.IOException: Cannot run program "{runtime.tools.dslite-9.2.0.1793-e1.path}/DebugServer/bin/DSLite": error=2, No such file or directory`

I am using the Linux version on 64 bit Ubuntu 18.04.

acidbourbon, 2020-07-24:
I have the exact same issue on Ubuntu 18.04 and Energia E23 ... i can has fix pls? :D :D

jackakyle, 2020-07-28:
Just coming back to this @acidbourbon, if you use the Boards Manager to update to V1.0.7 of the MSP430 boards it fixes this issue on E23.

Itamare4, 2020-07-31:
Ubuntu 18.04 - I have upgraded to 1.0.7, it worked according to @jackakyle advise - Thanks!

It stopped during the first firmware upgrade and now I get the following error - "One of the connected MSP-FETs / eZ-FETs debuggers needs recovery. Select 'Recover' to start the recovery process."