energia/Energia #1052: energia-1.8.10E23-macosx-signed fails to launch on macOS

rei-vilo, 2020-01-10:
Energia-1.8.10E23-macosx-signed fails to launch on macOS 10.15.2 _Catalina

[image]
_.

kelfink, 2020-01-15:
Same here, although I can't read French...  I'm running Mac OSX Mojave 10.14.6
[image]
Control-Click does let me "Show Package Contents" but I can't see what to run (if anything) from in there.

shuchow, 2020-01-20:
Same problem on 10.15.2

Initially, /var/log/system reported:

Jan 20 13:07:09 hostname com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x10000009.Energia[3884]): Could not find and/or execute program specified by service: 13: Permission denied: /Applications/Energia.app/Contents/MacOS/Energia

Permission was 0644.  I changed this to 777.  The app now immediately exists and this is written to system.log:

Jan 20 13:08:23 hostname com.apple.xpc.launchd[1] (nu.energia.Energia.24764[4325]): Service exited with abnormal code: 255

TianshuangQiu, 2021-02-01:
Wha

> Same problem on 10.15.2, but it looks like there's a workaround.
> 
> Initially, /var/log/system reported:
> 
> Jan 20 13:07:09 hostname com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x10000009.Energia[3884]): Could not find and/or execute program specified by service: 13: Permission denied: /Applications/Energia.app/Contents/MacOS/Energia
> 
> Permission was 0644. I changed this to 777, but the the app now immediately exists and this is written to system.log:
> 
> Jan 20 13:08:23 hostname com.apple.xpc.launchd[1] (nu.energia.Energia.24764[4325]): Service exited with abnormal code: 255
> 
> I then changed all of Applications/Energia.app/Contents to 777 and it launches. It's probably not a good idea. There's probably just one plist file or something that needs to be changed.

What do you mean change all to 777. Where can I change it?

elliott-turner, 2022-02-01:
> My error:
> 
> ```
> Exception Type:        EXC_BAD_ACCESS (SIGABRT)
> Exception Codes:       KERN_INVALID_ADDRESS at 0x00007fa7c1a18000
> Exception Note:        EXC_CORPSE_NOTIFY
> ```

I am consistently having the same issue with the latest release on macOS Big Sur. 1.8.7E21 also appears to suffer from the same issue, but 1.6.10E18 does not.

rei-vilo, 2022-02-01:
Unfortunately, as per #1065, Energia does not support macOS _Big Sur_.

elliott-turner, 2022-02-01:
@rei-vilo just seeing that now...

I now understand that Energia is not _expected_ to work on macOS _Big Sur_, but for anyone else who would like to _try_ and avoid having to downgrade their OS I was able to get it running by doing the following...

1. Download `energia-1.6.10E18-macosx-signed`
2. Launch Energia 1.6.10E18 (it should launch just fine, but you will notice that it fails to compile anything)
3. With Energia running (**do not quite the application**), right click on the Energia app and select "Show Package Contents"
4. Navigate to `Contents > Java > tools-builder > ctags`
5. Delete `5.8-arduino10`
6. Paste in `5.8-arduino11` (I just repeated step 4 but within the Arduino app instead of Energia and copied `5.8-arduino11` from there)
7. Energia should be able to compile and download sketches just fine now

Energia seems to work great now, even after closing and opening the application (I have not tested to see if it works after restarting my Mac however).

Unfortunately this sort of thing goes a bit above my head and I haven't spent a lot of time digging into it, but hopefully it might help someone else in my position.

rei-vilo, 2022-02-02:
@elliott-turner Thank you for the solution!