energia/Energia #1053: MSP430 on Energia 1.8.10E23 — Upload fails on macOS and Windows

rei-vilo, 2020-02-03:
Upload fails with Energia 1.8.10E23 on macOS when targeting the MSP430G2553. 

Energia generates 

```
~/Library/Energia15/packages/energia/tools/mspdebug/0.24/mspdebug rf2500 --force-reset prog '/var/folders/wn/n7qqb8ss0k3bvpqwfcdwp_7r0000gn/T/arduino_build_280546/Blink.ino.hex’ 
```

and it should be 

```
~/Library/Energia15/packages/energia/tools/mspdebug/0.24/mspdebug rf2500 --force-reset 'prog /var/folders/wn/n7qqb8ss0k3bvpqwfcdwp_7r0000gn/T/arduino_build_280546/Blink.ino.hex’
```

Note the `‘` before `prog`, not after.

The problem seems to appears on Windows too, as per https://forum.43oh.com/topic/19164-upload-error-filename-syntax-is-incorrect/?do=findComment&comment=90318

rei-vilo, 2020-02-03:
- Edit the file /Applications/Energia.app/Contents/Java/hardware/energia/msp430/platform.txt

- Change

```
tools.mspdebug.upload.pattern={cmd.path} {upload.verbose} {upload.protocol} --force-reset "prog '{build.path}/{build.project_name}.hex'"
```

for 

```
tools.mspdebug.upload.pattern={cmd.path} {upload.verbose} {upload.protocol} --force-reset 'prog "{build.path}/{build.project_name}.hex"'
```

Note the switch between single and double quotes.

I haven't tested this solution with paths that include spaces.

StefanSch, 2020-02-03:
I see some issues on Windows with this version
Suggest to use this one:
tools.mspdebug.upload.pattern={cmd.path} {program.verbose} {protocol} --force-reset "prog {build.path}/{build.project_name}.hex"

Note: should be changed for 
tools.mspdebug.program.pattern={cmd.path} {upload.verbose} {upload.protocol} --force-reset "prog {build.path}/{build.project_name}.hex"

as well.

Note : removed the ' and only use "
Can you check if this is working for OSX as well?

rei-vilo, 2020-02-03:
`tools.mspdebug.program.pattern={cmd.path} {program.verbose} {protocol} --force-reset "prog {build.path}/{build.project_name}.hex"` works for me on macOS.

I guess the second level of quotes was prompted by paths with spaces.

StefanSch, 2020-02-04:
yes, i also think this was the intention.
Just commited the fix.

closed 2020-02-04 by StefanSch