energia/Energia #1063: Allow Libraries to Extend Include Path

SciLor, 2020-07-27:
It would be helpful to be able to specify additional include path for libraries when they are compiled.

For ex. I want to port libopus for energia. To compile it I need to add include path for several subdirectories.

[See here](
https://github.com/arduino/arduino-cli/issues/501)

rei-vilo, 2020-07-27:
As the Energia IDE is based on Arduino, so the same limitations apply.

SciLor, 2020-07-27:
There is a pull request that addresses this issue.
https://github.com/arduino/arduino-cli/pull/502
May it be possible to use the arduino-builder from arduino?

Yes, I have tried to. The problem is that Enegia just takes the "library/src" folder as include.

rei-vilo, 2020-07-27:
This is consistent with 

> Maglie, C. et al. (2013, February-March). Arduino IDE 1.5: library specification. Retrieved from http://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification

You could either adapt the structure of the library or use another more capable IDE. 

Which board are you targeting, on which platform?

SciLor, 2020-07-27:
I am using a cc3200 which is used for a commercial product for what I am writing an alternative open source firmware. I have heavily modified/extended the cc3200 core for that.
I am developing on windows and Linux.

rei-vilo, 2020-07-27:
So you could use Code Composer Studio.

vChavezB, 2022-05-06:
Well, it seems that this issue will not be solved in the short term. I have developed a script that easily replaces include directives to relative ones. This should reduce the time it takes to port a 3rd party library to the Arduino ecosystem.

## Links
- [PyPi](https://pypi.org/project/relative-include)
- [Github](https://github.com/vChavezB/include-relative)

Based on this script I easily [ported ](https://github.com/vChavezB/lwip-Arduino) the lwIP TCP/IP stack into Arduino.