energia/Energia #1064: Target "${platform}-build" does not exist in the project "Energia" when building on Raspberry Pi 4

Pererva, 2020-08-03:
Hello!

I'm trying to build the Energia on Raspberry Pi 4 (version with 4Gb of RAM with the latest Rasbpian installed and all updates after `sudo apt-get update && sudo apt-get full-upgrade -y`).
I found [this old post](https://www.raspberrypi.org/forums/viewtopic.php?t=34790) on how to do that, but have no success when following it, even though it, basically, follows instructions on building Arduino IDE as specified in [ `howto.txt`] (https://github.com/energia/Energia/blob/master/build/howto.txt) file from repository.

### Used board
Launchpad with MSP430G2231 (marking on LaunchPad: MSP-EXP430G2)

### Steps to reproduce the problem:

Installing prerequisites:
```
# installing prerequisites
sudo apt-get install -y default-jdk ant texinfo libgmp-dev libmpfr-dev libmpc-dev
# building form GitHub repository
mkdir ~/"Electronics IDE"
cd ~/"Electronics IDE"
git clone --recursive https://github.com/energia/Energia.git
cd ./Energia/build
ant
```

### Result of actions:
```
Buildfile: /home/pi/Electronics IDE/Energia/build/build.xml

build:

BUILD FAILED
/home/pi/Electronics IDE/Energia/build/build.xml:82: The following error occurred while executing this line:
Target "${platform}-build" does not exist in the project "Energia".

Total time: 1 second
```

For a while I suspected that it didn't like the space in path, but using the folder without space in the path didn't change anything.
### Expected behaviour:
According to [this](http://code.google.com/p/arduino/wiki/BuildingArduino) : having no errors and being able to run `ant run` after build to use Energia

Is it fixable?

*P.S.* I also found [this post](http://shokai.org/blog/archives/tag/msp430launchpad) suggesting some libraries, which should be installed, but installing:
```
sudo apt-get install -y libusb-dev binutils-msp430 gcc-msp430 msp430mcu mspdebug msp430-libc gdb-msp430
```
didn't help to solve the issue. The error is still the same.

*P.P.S.*  The LaunchPad seems to be seen by RPi, according to [this post](https://labvolution.com/raspberry-pi-meets-ti-launchpad-msp430/) since after connecting them with USB cable and running `sudo mspdebug rf2500` I recieved the following reply:

```
MSPDebug version 0.22 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2013 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Trying to open interface 1 on 006
Initializing FET...
FET protocol version is 30001000
Set Vcc: 3000 mV
Configured for Spy-Bi-Wire
Device ID: 0xf201
	Code start address: 0xf800
	Code size         : 2048 byte = 2 kb
	RAM  start address: 0x200
	RAM  end   address: 0x27f
	RAM  size         : 128 byte = 0 kb
Device: MSP430F2012/G2231
...
Type "help <topic>" for more information.
Use the "opt" command ("help opt") to set options.
Press Ctrl+D to quit.
(mspdebug) 
```

rei-vilo, 2020-08-03:
Have you tried the revised procedure from https://github.com/robertinant/EnergiaNG#building-energia?

Pererva, 2020-08-03:
> Have you tried the revised procedure from https://github.com/robertinant/EnergiaNG#building-energia?

Yes. And it results in this error:
```
BUILD FAILED
/home/pi/Electronics IDE/EnergiaNG/build/build-energia.xml:136: The following error occurred while executing this line:
/home/pi/Electronics IDE/EnergiaNG/build/build-energia.xml:641: The following error occurred while executing this line:
/home/pi/Electronics IDE/EnergiaNG/build/build-energia.xml:214: The following error occurred while executing this line:
/home/pi/Electronics IDE/EnergiaNG/build/build-energia.xml:223: The following error occurred while executing this line:
/home/pi/Electronics IDE/EnergiaNG/build/build-energia.xml:837: Could not find file /home/pi/Electronics IDE/EnergiaNG/build/shared/reference.zip to generate checksum for.
```

MrityunjaiKumar, 2020-12-15:
I am facing the same error, and i suspect no download form url http://energia.nu/  is working. 
In the current log, build is trying to download http://energia.nu/files//reference.zip but not working.

Even https://energia.nu/download/ for pre build Energia is not working.