energia/Energia #1065: Support for macOS 11 Big Sur

rei-vilo, 2020-12-07:
Does Energia plan to support macOS 11 _Big Sur_?

rei-vilo, 2020-12-07:
Forget about macOS 11 _Big Sur_ as it introduces too many disruptions.

Keep macOS 10.15 _Catalina_ for stability and support of the LaunchPad and BoosterPack boards.

Previous macOS 10.14 _Mojave_ is even better as it runs 32-bit and 64-bit applications.

dybarsky, 2021-03-12:
Any plan to fix it? I am using platformio and there is same issue. I am blocked :( Probably I will just install linux on my macbook...

rei-vilo, 2021-03-12:
Shouldn’t Rosetta provide automatic support for Intel applications and run them seamlessly on M1?

dpharris, 2021-05-16:
Too late, Big Sur is installed ... am I to abandon Energia?

rei-vilo, 2021-05-16:
I am afraid I can't help you much, as I've switched to Linux.

Does the magic of Rosetta work?

nokonoko, 2021-06-20:
Support for Big Sur (intel), would be nice or at least so it doesn't crash when trying to start it. Is there any workaround?

dpharris, 2021-06-20:
My solution was to revert to Catalina.  My older macbookpro was randomly
crashing on Big Sur, and then continued to do so on Catalina, until I
disabled one of the Thunderbolt kexts.  A huge time waste.

David

rei-vilo, 2021-06-20:
I have taken an even more radical approach: I switched to Linux. 

Debian / Gnome is stable and light, and Visual studio Code works very well. 

No regrets.

dpharris, 2021-06-20:
I will likely reactivate my virtualbox linux.   Unfortunately, I also have
to have a vb Windows since some of my software doesn't have osx/linux
versions!?

David

dybarsky, 2021-06-21:
I have installed Linux mint on my macbook and this solves a lot of issues for me. Apple doesn't produce good reliable software anymore.

dpharris, 2021-06-21:
> To be honest after all updates on Big Sur recently my Macbook doesn't
> crash or act up like it did in the beginning. Downgrading to Catalina or
> switching to Linux just for one program seems rather dramatic. But each to
> their own I guess.

True enough.  It is not just one program, though.  And the crashing made
the macbook unusable, despite the latest updates up to a few weeks ago
(applying those updates was incredibly hard, too, as it kept crashing ---
single user mode to the rescue, at a performance hit.  Perhaps the updates
in the last few weeks fixed the problem?  At this point I am: once bit,
twice shy.

David

nokonoko, 2021-06-21:
Understandable if it crashes that often, for me it was around once a week until 3 months or so when the updates fixed something. I do agree that Apple shouldn't have rushed Big Sur so much since new minor problems keep appearing for people, at least they release updates to fix them which is nice... I guess.

dpharris, 2021-06-21:
I am not convinced Apple has fixed the problems .. at least for older
macbooks.  Many complaints out there.  Probably just time for one of those
new M1 Airs.

David

rei-vilo, 2021-06-21:
The major issues I faced were 

* the drivers still 32-bit,
* the system extensions (kext) no longer supported by the new release of macOS,
* hardware manufacturers no longer supporting macOS.

nokonoko, 2021-06-21:
Still rocking a 2014 MacBook Pro so I think y'all very unlucky.

rei-vilo, 2022-02-01:
I was surprised how nicely Windows Subsystem for Linux (WSL) works.

rei-vilo, 2022-02-02:
Find a solution for _macOS_ Big Sur at https://github.com/energia/Energia/issues/1052#issuecomment-1027390752