energia/Energia #1067: Boards Manager cannot download board packages

commandblockguy, 2021-02-16:
### Please provide the following information.

* LaunchPad board name and version: `EK-TM4C123GXL`
* Energia IDE version (found in Energia > About Energia menu): `1.8.11E23`
* Board package version (found in Tools > Boards > Boards Manager menu): `(not installed yet)`
* OS name and version: `Linux Mint 20.1`

### Describe the expected results.

After clicking the install button for the TivaC boards, the package for TivaC boards would be installed.

### Describe the actual results.

The package is not installed, and the board manager gives the error `Error downloading `http://energia.nu/downloads/download_core.php?file=tivac-1.0.4.tar.bz2`. The following is printed to the console:
```
2021-02-16T00:47:24.172Z INFO c.a.u.n.HttpConnectionManager:148 [ContributionManager Install Thread] Connect to http://energia.nu/downloads/download_core.php?file=tivac-1.0.4.tar.bz2, method=GET, request id=B408A3349F564EE8
2021-02-16T00:47:24.329Z INFO c.a.u.n.HttpConnectionManager:152 [ContributionManager Install Thread] Request complete URL="http://energia.nu/downloads/download_core.php?file=tivac-1.0.4.tar.bz2", method=GET, response code=302, request id=B408A3349F564EE8, headers={Keep-Alive=[timeout=5], null=[HTTP/1.1 302 Found], Server=[Apache], Connection=[Keep-Alive], Content-Length=[255], Date=[Tue, 16 Feb 2021 00:47:24 GMT], Content-Type=[text/html; charset=iso-8859-1], Location=[https://energia.nu/downloads/download_core.php?file=tivac-1.0.4.tar.bz2]}
2021-02-16T00:47:24.330Z INFO c.a.u.n.HttpConnectionManager:159 [ContributionManager Install Thread] The response code was a 301,302 so try again with the new URL https://energia.nu/downloads/download_core.php?file=tivac-1.0.4.tar.bz2
2021-02-16T00:47:24.333Z INFO c.a.u.n.HttpConnectionManager:148 [ContributionManager Install Thread] Connect to https://energia.nu/downloads/download_core.php?file=tivac-1.0.4.tar.bz2, method=GET, request id=555480785CD24890
2021-02-16T00:47:24.582Z INFO c.a.u.n.HttpConnectionManager:152 [ContributionManager Install Thread] Request complete URL="https://energia.nu/downloads/download_core.php?file=tivac-1.0.4.tar.bz2", method=GET, response code=500, request id=555480785CD24890, headers={null=[HTTP/1.1 500 Internal Server Error], Server=[Apache], Connection=[Upgrade, close], Vary=[User-Agent], Content-Length=[0], Upgrade=[h2,h2c], Date=[Tue, 16 Feb 2021 00:47:24 GMT], Content-Type=[text/html; charset=UTF-8], X-Powered-By=[PHP/7.3.23]}
2021-02-16T00:47:24.583Z ERROR c.a.u.n.FileDownloader:199 [ContributionManager Install Thread] The request stop
java.io.IOException: Received invalid http status code from server: 500
	at cc.arduino.utils.network.FileDownloader.openConnectionAndFillTheFile(FileDownloader.java:239) ~[arduino-core.jar:?]
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:182) ~[arduino-core.jar:?]
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:129) ~[arduino-core.jar:?]
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:142) ~[arduino-core.jar:?]
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:78) ~[arduino-core.jar:?]
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60) ~[arduino-core.jar:?]
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:105) ~[arduino-core.jar:?]
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172) ~[pde.jar:?]
	at java.lang.Thread.run(Thread.java:748) [?:1.8.0_231]
 Error downloading http://energia.nu/downloads/download_core.php?file=tivac-1.0.4.tar.bz2
java.lang.RuntimeException: java.lang.Exception: Error downloading http://energia.nu/downloads/download_core.php?file=tivac-1.0.4.tar.bz2
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading http://energia.nu/downloads/download_core.php?file=tivac-1.0.4.tar.bz2
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:144)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:78)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:105)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 500
	at cc.arduino.utils.network.FileDownloader.openConnectionAndFillTheFile(FileDownloader.java:239)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:182)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:129)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:142)
	... 5 more
```

### List the steps to reproduce the problem.

Open the boards manager, and click the install button for any package.

robertinant, 2021-02-16:
Fixed. Database was down.

robertinant, 2021-02-16:
Thank you all for reporting this. Appreciate it!

closed 2021-02-16 by robertinant